JuliaRobotics/DistributedFactorGraphs.jl #17: Performance figures on Julia Dict{Symbol, T} lookups

dehann, 2019-05-31:
Just pasting here for reference, but looks like Julia 1.1 at least is pretty darn fast.

```julia
julia> for i in round.(Int,1000000*rand(10000))
         d[Symbol("x$i")] = randn(3,3)
       end

julia> length(d)
9956

julia> @benchmark d[:x972233]
BenchmarkTools.Trial: 
  memory estimate:  0 bytes
  allocs estimate:  0
  --------------
  minimum time:     19.154 ns (0.00% GC)
  median time:      19.410 ns (0.00% GC)
  mean time:        19.822 ns (0.00% GC)
  maximum time:     43.192 ns (0.00% GC)
  --------------
  samples:          10000
  evals/sample:     997

julia> @benchmark d[:x748125]
BenchmarkTools.Trial: 
  memory estimate:  0 bytes
  allocs estimate:  0
  --------------
  minimum time:     20.230 ns (0.00% GC)
  median time:      20.683 ns (0.00% GC)
  mean time:        21.467 ns (0.00% GC)
  maximum time:     57.373 ns (0.00% GC)
  --------------
  samples:          10000
  evals/sample:     997

julia> @benchmark d[:x748125]
BenchmarkTools.Trial: 
  memory estimate:  0 bytes
  allocs estimate:  0
  --------------
  minimum time:     20.070 ns (0.00% GC)
  median time:      20.267 ns (0.00% GC)
  mean time:        20.867 ns (0.00% GC)
  maximum time:     55.548 ns (0.00% GC)
  --------------
  samples:          10000
  evals/sample:     997
```

System info:
```julia
julia> versioninfo()
Julia Version 1.1.1
Commit 55e36cc308 (2019-05-16 04:10 UTC)
Platform Info:
  OS: Linux (x86_64-pc-linux-gnu)
  CPU: Intel(R) Core(TM) i7-7500U CPU @ 2.70GHz
  WORD_SIZE: 64
  LIBM: libopenlibm
  LLVM: libLLVM-6.0.1 (ORCJIT, skylake)
Environment:
  JULIA_NUM_THREADS = 4
```

cc @GearsAD -- looking pretty good!

dehann, 2019-05-31:
cc @Affie

GearsAD, 2019-06-01:
#symbolsforlife!

dehann, 2019-06-02:
My concern now is that the tests above are with regard to a static defintion of the lookup variable.  Should probably be done as a function with new random variable selection.

dehann, 2019-06-08:
I redid this with a randomized lookup:
```julia
julia> lookup = [:x421546; :x3891; :x905913]

julia> getid() = round(Int, 2*rand()+1)

julia> @benchmark d[lookup[getid()]]
BenchmarkTools.Trial: 
  memory estimate:  0 bytes
  allocs estimate:  0
  --------------
  minimum time:     34.326 ns (0.00% GC)
  median time:      35.920 ns (0.00% GC)
  mean time:        36.430 ns (0.00% GC)
  maximum time:     104.083 ns (0.00% GC)
  --------------
  samples:          10000
  evals/sample:     993
```

It's till pretty fast!  I'm not sure why my test with Graphs.jl ended up being 1000x slower when I switched the incdict to be `Symbol` rather than `Int`.  Hopefully just a fluke or something silly resulting in a lot of unnecessary conversions.  Will keep testing this ince the new code stabilized with backup plan of using temporary internal `Int` ids, but doesn't look necessary at this point.

closed 2019-06-08 by dehann