JuliaRobotics/DistributedFactorGraphs.jl #20: DeleteVariable should clean up _variableOrderSymbols in neighbor factors

GearsAD, 2019-06-01:


Affie, 2020-02-07:
I hope I understand `_variableOrderSymbols` correctly. 
From how I see it, it is attached to a factor's definition and should not be modified at all.
How else would you know a factor is orphaned and does not have all the variables associated with it?
Its important in IIF also and used there as:
`getVariableOrder(fct::DFGFactor)::Vector{Symbol} = fct._variableOrderSymbols`

dehann, 2020-02-12:
at least one major use is that factors are order sensitive to the variables being passed in and this is where that info is stored

Affie, 2020-02-12:
> DeleteVariable should clean up _variableOrderSymbols in neighbor factors

I would say a definite **no** on this, and close issue?

Affie, 2020-02-12:
Sorry didn't mean to close

dehann, 2020-02-12:
umm i think it’s a definite yes — perhaps we thinking of different things.  if you delete a variable, then i can not possibly exist in a factor lookup?  means the factor is incomplete and should fail if evaluated.  default should probably be to also delete factors attached to the deleted variable too, no?

dehann, 2020-02-12:
As usual, @Affie convinced me to NOT delete from `_variableOrderSymbols`, and that the error should be found by checking the current neighbors to a factor.

IF a variable is deleted from DFG object, then default is to delete surrounding factors also.  This can be avoided with a keyword argument.

If a factor is not deleted and a new variable is introduced, then a new factor must be created by one of the verbs and replace the current factor which has (by this time) a stale but immutable `_variableOrderList`.

One reason is that deleting and adding variables will not preserve the all important order.  Another reason is that deleting both the edges in the graph and variables from this list undermines the test where one compares the two lists against each other -- currently used for orphan test.  Third reason is when using the `multhypo` interface, then the number of variables is specific to that instance of the factor (as in all cases really) and therefore `_variableOrderList` must be immutable.

Affie, 2020-02-13:
Close in favor of #296

closed 2020-02-13 by Affie