JuliaRobotics/DistributedFactorGraphs.jl #22: which neighbors method to use regardless of ready or backendset?

dehann, 2019-06-03:
Hi, which function should I use to get neighbors regardless of their internal state?  I used to use `ls` -- I see the current `getNeighbors` is internal state specific?

dehann, 2019-06-03:
- when looking for factors around a variable, it looks like `lsf` can maybe still do the job?  
- does that mean `ls` for variables around a factor?
- I'm curious about the general case too though?

GearsAD, 2019-06-03:
> Hi, which function should I use to get neighbors regardless of their internal state? I used to use `ls` -- I see the current `getNeighbors` is internal state specific?

Hi, if you pass `backendset` and `ready` as nothing (or leave as default), getNeighbors won't filter.

GearsAD, 2019-06-03:
Looking at the `lsf` you added, it seems that `ls(dfg, :x1)` and `lsf(dfg, :x1` have the same behavior. I made `ls` and `lsf` for more listing the nodes in the graph, not their relationships.

I would suggest:
- Use `ls(dfg)::Vecror{DFGVariable}` or `getVariableIds(dfg)::Vector{Symbol}` for all variables in the graph (or regex a set)
- Use `lsf(dfg)::Vector{DFGFactor}` or `getFactorIds(dfg)::Vector{Symbol}` for all factors in the graph (or regex a set)
- Use `ls(dfg, symbol)::Vector{Symbol}` for neighbors irrespective of variable or factor  

We can refactor for the correct return types - I see that it's not evident what the return type is.

dehann, 2019-06-03:
### Maintenance

>  it seems that ls(dfg, :x1) and lsf(dfg, :x1 have the same behavior

Cool, we can concentrate these too if that makes more sense.

### Decision

> We can refactor for the correct return types

I'm not sure what the right way would be but would suggest we imagine / ask around on what the "expected" API behaviour would be... 
- So there are two return types, either `::Symbol` or `::DFGVariable/::DFGFactor`.
- There are roughly three functions: `getNeighbors`, `ls/lsf`, and `getVariableIds/getFactorIds`.

> **Targeting** IIF v0.8.x

### Suggestion Rev.1

My knee jerk reaction is to:
- take inspiration from bash `ls` and `ls -hla`, but also 
- solve known performance bottleneck (no issue found, occurred during Arena.jl development) on wanting to return values and labels from a server.
- should be thinking about bash `cat somefile` as inspiration too (can always alias)?

Therefore,
- `ls/lsf/ls2` is shallow -- i.e. return `::Vector{Symbol}`,
  - Maybe something like `ls(dfg, summary=true)` return `::Tuple{Symbol, type, MAP_est, ...}`.
- `ls(dfg, :x13) / lsf(..) / ls2(..)` is shallow -- i.e. return `::Vector{Symbol}`,
  - Maybe something like `ls(dfg, :x13, summary=true)` return `::Tuple{Symbol, type, MAP_est, ...}`.
- `getVariableIds/getFactorIds(dfg, lbl::Symbol)::Vector{Symbol}` just the same as `ls\lsf`.
- `getNeighbors(dfg, sym)` always returns either `::Union{Vector{DFGVariable}, Vector{DFGFactor}}`,
  - For example `getNeighbors(dfg, :x13)::Vector{DFGFactor}`, or  `getNeighbors(dfg, :x1x2f1)::Vector{DFGVariable}`
  - This implies there is a 'cross-over' from request `::Symbol` and get `::DFGVariable/DFGFactor`.
    - What about the return 'cross-over' from `DFGVariable/DFGFactor` to `::Symbol`?

dehann, 2019-06-03:
cc @Affie , @pvazteixeira , @nicrip -- any comments/suggestions/preferences on this perhaps please?

GearsAD, 2019-06-19:
No news is good news? :) I suggest we move ahead with #40 as the fix and close this?

dehann, 2019-06-21:
Hi, yeah I think #40 is a good way to go.  I added more comments there.

dehann, 2019-06-21:
I'd suggest using `getVariables(fg)::Vector{DFGVariable}` or `getFactors(fg)::Vector{DFGFactor}` which I believe was the previous behaviour of `DFG.ls(fg)`.  I personally prefer `ls(fg)::Vector{Symbol}` as suggested in #40, then `ls` and `lsf` always return `::Vector{Symbol}`.

closed 2019-06-21 by dehann

dehann, 2019-06-21:
xref JuliaRobotics/DistributedFactorGraphs.jl#304