JuliaRobotics/DistributedFactorGraphs.jl #29: Make DFG type stable

dehann, 2019-06-09:
Maybe target DFG v0.2.x to not have dynamic types such as `Union` or `Any`, for example here:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/aa7ac5df8cec93d134cd43c453949aecd663461e/src/services/GraphsDFG.jl#L54

See here:
https://docs.julialang.org/en/v1/manual/performance-tips/#Avoid-containers-with-abstract-type-parameters-1

dehann, 2019-06-09:
If solveparams becomes an abstract type, we should just use templating for performance:
```julia
mutable struct MyType{T}
  solverParams::T
end
```

dehann, 2019-06-09:
In that case:
```julia
mutable struct GraphsDFG{T} <: AbstractDFG
    g::FGType
    description::String
    nodeCounter::Int64
    labelDict::Dict{Symbol, Int64}
    addHistory::Vector{Symbol} #TODO: Discuss more - is this an audit trail?
    solverParams::T # Solver parameters
end
```

GearsAD, 2019-06-09:
Sounds like a plan! Yep, it was just a placeholder, would be good to template that part

GearsAD, 2019-06-09:
There's no inheritance, it would be nice to have that in AbstractDFG, but we'll just standardize a solverParms in every implementation of AbstractDFG.

dehann, 2019-06-10:
still need to do #33 , but actually mostly finished the work with #31 already.

closed 2019-06-10 by dehann