JuliaRobotics/DistributedFactorGraphs.jl #36: what to do with `_evalType` and several other encodings currently in IIF

dehann, 2019-06-10:
Trying to concentrate the pickling/unpickling, serialization/deserialization, packing/unpacking, encoding/decoding functions:

- [ ] [`IIF/evalType`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/5650955bda454251c1855dd0380e99ee249b78fc/src/FGOSUtils.jl#L506)
- [ ] [`IIF/getpackedtype`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/5650955bda454251c1855dd0380e99ee249b78fc/src/DispatchPackedConversions.jl#L91)
- [ ] [`IIF/convert2packed...`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/5650955bda454251c1855dd0380e99ee249b78fc/src/DispatchPackedConversions.jl#L121)
- [ ] [`IIF/encodefg`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/5650955bda454251c1855dd0380e99ee249b78fc/src/DispatchPackedConversions.jl#L154)
- [ ] convert methods such as for example [`IIF/convert(::Type{Packed...}...`](https://github.com/JuliaRobotics/IncrementalInference.jl/blob/5650955bda454251c1855dd0380e99ee249b78fc/src/DispatchPackedConversions.jl#L18)

GearsAD, 2019-06-19:
In CloudgraphsDFG I was able to get by only using `IncrementalInference.decodePackedType`. Let me check on whether we can do an end-to-end solve with the CG driver, if so, I'd like to remove all the extraneous functions from IIF.

dehann, 2020-04-27:
cc @Affie

Affie, 2020-05-20:
Can be removed: 
- `IIF/encodefg` already removed and just a comment.
- `IIF/convert2packed...` not used anywhere
- `IIF/getpackedtype` looks like just an export without any functions
- `IIF/evalType` In deprecated and looks like its not called

Affie, 2020-05-20:
> convert methods such as for example IIF/convert(::Type{Packed...}...

This is the core that is needed and has to be provided for every type.

dehann, 2020-05-24:
must add this to docs:
https://juliarobotics.org/Caesar.jl/latest/concepts/adding_variables_factors/

we are already doing this for all types in IIF/RoME/Caesar :-)

closed 2020-05-24 by dehann

dehann, 2020-08-05:
`_evalType` is still used by ZMQ interface in Caesar.  This should be updated toward a standardized MsgEndpoint interface.