JuliaRobotics/DistributedFactorGraphs.jl #38: Consider introducing JuliaDB_DFG for locally storing dfg objects and data to file

dehann, 2019-06-13:
https://juliadb.org/

dehann, 2019-06-13:
this would replace the current and wonky JLD2 approach currently used by IIF.

dehann, 2019-06-13:
Maybe also worth thinking about SQLite for wider support, although that might just be too general and too weird in other languages to be useful?  It is unlikely that the solver will be moved to another language and more likely that we would stay with the [ZMQ server interface](https://github.com/JuliaRobotics/Caesar.jl/blob/c8b58a14600bd91dfbee992315923722d264b1b5/src/zmq/ZmqCaesar.jl#L1) instead.

dehann, 2019-06-14:
cc @andreasnoack, we just started this discussion on whether to use JuliaDB for file storage of our factor graph objects.  Might take some time to fully flesh this out but thought you might like to see some of the discussion when it happens.  The idea would be to wrap a "file" or even entire folder-of-files to persist the factor graph with an interface similar to:
- [Graphs.jl based](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/3bce87fb910166b3527e61d564cd06c4389d5467/src/GraphsDFG/GraphsDFG.jl#L1)
- [CloudGraphs.jl based](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/3bce87fb910166b3527e61d564cd06c4389d5467/src/CloudGraphsDFG/CloudGraphsDFG.jl#L1)

GearsAD, 2019-06-15:
Yes, good topic! I think we have the encoding figured out, we need to consider how/where to store the structure and the relationships. More to follow on this once CloudGraphs.jl is completed.

dehann, 2019-06-17:
xref #42

dehann, 2019-07-08:
cc @pvazteixeira since ZMQ work and serialization have a lot of overlap.

closed 2019-07-22 by dehann