JuliaRobotics/DistributedFactorGraphs.jl #42: Investigate implementing a Fuse DFG driver

GearsAD, 2019-06-16:
As a user I would like a virtual file system (with reading+writing) for all graphs. 

Details:
- Consider using GraffAPI so that we can embed auth+auth into it
- Consider ZMQ and try align the two so they share a common set of contracts
- Consider big data and how to read+write it 

A Wiki page and a small PoC is sufficient for this issue.

Ref: [Fuse C example](https://github.com/libfuse/libfuse/blob/master/example/hello.c)

dehann, 2019-06-16:
One thing to remember is that the libfuse process disappears from userspace processes.  Just to help decide, should the fuse driver not perhaps be a second process that talks via shared memory or IPC using ZMQ to the big SLAM process?

GearsAD, 2019-06-16:
That would be different design then - where the fuse directly interacts with a SLAM system serially. I was thinking more of decoupled interaction (via the graph). Good to discuss, depends on what the user wants to do with it - 
- If it's for data ingestion and extraction, I would suggest the latter
- If it's for interacting with a solver, the former (it would be a richer driver with all the ZMQ bells+whistles but would be more complex).

dehann, 2019-06-16:
Not sure fuse drops all the way to a kernel space process, but either way I’m fairly sure we don’t want to drop the full SLAM processes/threads out from user space into the invisible OS processes.  Will have to investigate more about what happens to a fuse process to be sure.  Yes, going through zmq is more complicated.  

I’m assuming the FuseDFG driver would give the user a serialized data stream?

dehann, 2019-06-17:
influences #38 decision making