JuliaRobotics/DistributedFactorGraphs.jl #44: Clean up serialization module code

GearsAD, 2019-06-18:
How should we do this?

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/master/src/services/AbstractDFG.jl#L19-L26

dehann, 2019-06-21:
Oh, yeah that ... will have to think on that a bit more.  Shall we target DFG v0.2.0 for a decision here?

dehann, 2019-06-21:
Ideally we don't want to have to "register" anything and just allow dispatch to find the required converters.  I recall we were trying to make this work without and weren't able to resolve it at the time.  I think the issue was module name discovery for some reason.  Looks like you always just consider `Main` which feels like the right way to do it -- think that is further than what I could get working previously.

Affie, 2020-05-01:
@GearsAD or @dehann, with reference to what is this? the code link is broken

Affie, 2020-05-17:
It was getSerializationModule

closed 2020-05-17 by Affie