JuliaRobotics/DistributedFactorGraphs.jl #45: tag v0.1.3

dehann, 2019-06-21:
@JuliaRegistrator register()

JuliaRegistrator, 2019-06-21:
Registration pull request created: [JuliaRegistries/General/1489](https://github.com/JuliaRegistries/General/pull/1489)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.3 -m "<description of version>" ba79e8bd3be6c892f7a47cac7d45fec1b16f5e3d
git push origin v0.1.3
```

dehann, 2019-06-21:
@JuliaRegistrator register()

JuliaRegistrator, 2019-06-21:
Registration pull request updated: [JuliaRegistries/General/1489](https://github.com/JuliaRegistries/General/pull/1489)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.3 -m "<description of version>" 0b182a204cfafc1a93e153f00e279b7bcf54a30a
git push origin v0.1.3
```

closed 2019-06-22 by dehann