JuliaRobotics/DistributedFactorGraphs.jl #58: FEATURE lsWho, lsfWho

dehann, 2019-07-09:
Add methods that gives back all factor labels that fit the bill:
- [ ] `lsfWho(fg, :Point2Point2)`
- [ ] `lsWho(fg, :Pose3)`

return type `::Vector{Symbol}`

dehann, 2019-07-09:
hey @nicrip, this might be a fun one to try early on.

dehann, 2019-07-09:
complement to:
[image]

Affie, 2019-08-22:
@dehann I quickly added something for this: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/0a1053483705a2f33f5cb3bf08709e64e37d1089
Is it what you had in mind?

dehann, 2019-08-25:
HI @Affie, yes that is great thanks!  We could also do with the types directly (in addition to Symbol):
```julia
lsWho(fg, Pose3)
```
Did you want to merge the jt/deveop branch at some point or cherry pick a separate PR from it?  I'm happy to merge stuff, as long as tests do a good job.

Affie, 2019-08-26:
Ok, that shouldn't be a problem to do. Its a bit harder to test though. 
If you want it earlier we can cherry pick, it is mostly separate commits.
The develop branch has breaking changes and I had it in mind as v0.4.0. I just need some input from you and Sam.

dehann, 2019-09-21:
We could even go as far as:
```julia
ls(fg, Pose2Point2BearingRange)
lsf(fg, Pose2Point2BearingRange)

ls(fg, Pose2)
```

closed 2019-09-22 by dehann