JuliaRobotics/DistributedFactorGraphs.jl #95: saveDFG / loadDFG does not preserve order in .fncargvID field

dehann, 2019-09-03:
see JuliaRobotics/Caesar.jl#382

GearsAD, 2019-09-09:
After testing this (unable to reproduce in a test graph), and then looking through the code, it seems that this _should_ be working as expected - https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/v0.3.2/src/FileDFG/services/FileDFG.jl#L101-L103

Can I look at the specific example?

dehann, 2019-09-09:
yeah, upon closer inspection i see that this might also occur in function `buildSubgraphFromLabels`.  Haven't dug any deeper than that yet.

dehann, 2019-09-09:
definitely in `buildSubgraphFromLabels` even before the test data was saved with `saveDFG`

dehann, 2019-09-09:
xref JuliaRobotics/IncrementalInference.jl#354

GearsAD, 2019-09-09:
Preliminary fix: `https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/tree/hotfix/v0_3_2_orderingissue`

Affie, 2019-09-09:
It looks like Light, Symbol DFG does not have this error. I added a test for this https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/fedcfceeeb35e797003d0847922c563d58c7be77

GearsAD, 2019-09-09:
Tagged 0.3.3 with a fix.

closed 2019-09-09 by GearsAD