JuliaRobotics/DistributedFactorGraphs.jl #96: IIF as a test dependancy

Affie, 2019-09-05:
Bringing up #75 again. We discussed to **temporary** bring IIF back into DFG test dependencies. It looks like its currently only needed for serialization, until a better solution can be found. The reasoning is rather have test with downstream dependency than no tests at all. Can help avoid issues such as #95.
@dehann

Affie, 2019-09-06:
RoME is also needed for types such as `Pose2`

dehann, 2019-09-09:
Hi, is there any hope to cut out the offending part of IIF and have that in DFG, or is it too big a piece.  The history here is that everything was in in IIF and then DFG came along and some stuff has migrated upstream there.  If this test only requires like one or two functions from IIF, there might be a case to just upstream that functionality.

Broad focus areas are:
- DFG is (not just a wrapper for many graphing technologies but instead) a surgical focus package for working with distributed factor graphs that depend on various graphing technologies.
- IIF is algebraic structure for inference, and tries to be agnostic,
- RoME builds on the transform and manifold layers for factors and variables specific to robotics.

GearsAD, 2019-09-09:
This is just for testing and it may be difficult to move everything across for a decent test. My opinion is that it's a lesser evil to bring downstream in for testing than to not have the tests. Let's see how it plays out in the next few days of changes though?

Affie, 2019-09-10:
I propose that we start off with RoME and IIF for now. 
- As the tests matures we can start seeing what exactly is needed and move it to tests/DFG.
- What about creating a big beehive data set either in code or saved to a file (fileDFG?). 
- Only DFG specific test should be done

The downside is that as soon as a new DFG version is tagged it will break the build until IIF and RoME is updated. eg. going from v0.3.2 to v0.4.0 if IIF depends on ^v0.3.2.

GearsAD, 2019-09-20:
Yeah the circular dependencies are going to be a headache. I will try see if i can move enough to not have to import IIF/RoME

GearsAD, 2019-10-17:
In #132, when merged we can finally close this.

closed 2019-10-22 by Affie