JuliaRobotics/DistributedFactorGraphs.jl #97: Add big data

Affie, 2019-09-05:


Affie, 2019-10-03:
Store keys=>AbstractBigData vs. only keys
 
keys=>AbstractBigData
- To guarantee small payload: 
  - only dispatch to cloud on specific <:AbstractBigDataElement types. 
  - In memory (should) pass pointers around so it should ideally not matter.
- On delete variable: big data entry should be deleted externally

Affie, 2019-10-04:
WIP PR https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/131

GearsAD, 2019-12-01:
Updated in #228 using your AbstractBigDataPattern

GearsAD, 2019-12-01:
AbstractBigDataEntry looks good - added a `GeneralBigDataEntry`, should support general use-cases - https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/228/files#diff-a9bf58a6f5bf50be944b4eb28d78862eR6-R12

Affie, 2020-05-01:
Confirm if this is in CGDFG

closed 2020-08-12 by Affie