JuliaRobotics/DistributedFactorGraphs.jl #110: loadDFG, 'Prior', please check the unpacking/packing converters for this factor

dehann, 2019-09-13:
```
ERROR: Error while unpacking 'x0f1' as 'Prior', please check the unpacking/packing converters for this factor - 
type GraphsDFG has no field decodePackedType
Stacktrace:
 [1] getproperty(::Any, ::Symbol) at ./sysimg.jl:18
 [2] _unpackFactor(::DistributedFactorGraphs.GraphsDFG{DistributedFactorGraphs.NoSolverParams}, ::Dict{String,Any}, ::DistributedFactorGraphs.GraphsDFG{SolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:79
 [3] loadDFG(::String, ::DistributedFactorGraphs.GraphsDFG{SolverParams}, ::DistributedFactorGraphs.GraphsDFG{DistributedFactorGraphs.NoSolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:165
 [4] loadDFG(::String, ::DistributedFactorGraphs.GraphsDFG{SolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:141
 [5] top-level scope at none:0
 [6] eval(::Module, ::Any) at ./boot.jl:328
 [7] eval_user_input(::Any, ::REPL.REPLBackend) at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.1/REPL/src/REPL.jl:85
 [8] macro expansion at /buildworker/worker/package_linux64/build/usr/share/julia/stdlib/v1.1/REPL/src/REPL.jl:117 [inlined]
 [9] (::getfield(REPL, Symbol("##26#27")){REPL.REPLBackend})() at ./task.jl:259
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] _unpackFactor(::DistributedFactorGraphs.GraphsDFG{DistributedFactorGraphs.NoSolverParams}, ::Dict{String,Any}, ::DistributedFactorGraphs.GraphsDFG{SolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:85
 [3] loadDFG(::String, ::DistributedFactorGraphs.GraphsDFG{SolverParams}, ::DistributedFactorGraphs.GraphsDFG{DistributedFactorGraphs.NoSolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:165
 [4] loadDFG(::String, ::DistributedFactorGraphs.GraphsDFG{SolverParams}) at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:141
 [5] top-level scope at none:0
```

cc @Affie SB-210c8c

dehann, 2019-09-13:
oh nevermind, sorry.

dehann, 2019-09-14:
happened again in #lalala, should probably do better error msg.

GearsAD, 2019-09-23:
Renaming this

dehann, 2019-10-21:
renaming to what if you dont mind me asking? :-)

GearsAD, 2019-10-21:
Function signature changed in #171 to:
```{julia}
function loadDFG(folder::String, iifModule, dfgLoadInto::G) where G <: AbstractDFG
```

dehann, 2019-10-21:
why iifModule?  the converters can be defined anywhere right, many of them in RoME currently, but will always be near the top and user...  Is that okay?

dehann, 2019-10-21:
Another suggestion (sorry know its annoying)  static parameter over use thing:
```julia
function loadDFG(folder::AbstractString, iifModule, dfgLoadInto::AbstractDFG)
```

dehann, 2019-10-21:
the intended use is then (just making sure) -- something like:
```julia
fg = initfg()
loadDFG(mypath[2:end], Main, fg)
```

GearsAD, 2019-10-21:
Yep, that work? Otherwise we have to assume the type of graph your want it to create. Open to suggestions

dehann, 2019-10-21:
thats perfect in my opinion, ship it make millions of units!  what could possibly go wrong.

dehann, 2019-10-21:
nothing more we can do about that right now, may need to change when we figure out how to avoid the `Main` call.  Thanks!

Affie, 2020-05-01:
This was fixed some time ago

closed 2020-05-01 by Affie