JuliaRobotics/DistributedFactorGraphs.jl #111: Lightgraphs error?

GearsAD, 2019-09-20:
┌ Error: Depreciated? Concider using exists(dfg,label) instead. Returing internals copy
└ @ DistributedFactorGraphs.LightDFGs ~/.julia/dev/DistributedFactorGraphs/src/LightDFG/entities/LightDFG.jl:40
stacktrace() = Base.StackTraces.StackFrame[macro expansion at show.jl:555 [inlined], getproperty(::LightDFG{NoSolverParams,DFGVariable,DFGFactor}, ::Symbol) at LightDFG.jl:41, getLabelDict(::LightDFG{NoSolverParams,DFGVariable,DFGFactor}) at LightDFG.jl:4, top-level scope at interfaceTests.jl:59, top-level scope at Test.jl:1083, top-level scope at interfaceTests.jl:43, include at boot.jl:326 [inlined], include_relative(::Module, ::String) at loading.jl:1038, include(::Module, ::String) at sysimg.jl:29, include(::String) at client.jl:403, top-level scope at runtests.jl:9, top-level scope at Test.jl:1083, top-level scope at runtests.jl:8, include at boot.jl:326 [inlined], include_relative(::Module, ::String) at loading.jl:1038, include(::Module, ::String) at sysimg.jl:29, include(::String) at client.jl:403, top-level scope at none:0, eval(::Module, ::Any) at boot.jl:328, exec_options(::JLOptions) at client.jl:243, _start() at client.jl:436]

GearsAD, 2019-09-20:
Looks to be by design, TBD.

Affie, 2019-09-26:
Didn't want to create (and keep updated) a field that was not needed. It currently points to the internals, but it can be dangerous to directly use it as it is not fixed. The though was to replace it with function calls such as `exists`. I still need to trace all the calls to it to make sure it is not important.

Affie, 2019-09-27:
Can we close this now with #122 merged?

GearsAD, 2019-09-28:
Yep, thanks

closed 2019-09-28 by GearsAD