JuliaRobotics/DistributedFactorGraphs.jl #118: Release v0.4 (Prerelease)

GearsAD, 2019-09-26:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2019-09-26:
Registration pull request created: [JuliaRegistries/General/3837](https://github.com/JuliaRegistries/General/pull/3837)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.0 -m "<description of version>" 735e8ec6c3cf23a46a45e622db4ee45ef7a411f2
git push origin v0.4.0
```

dehann, 2019-09-27:
oh Sam, this is not going to work -- need to bump on registries.

GearsAD, 2019-09-27:
Ah, haha yeah

closed 2019-09-28 by dehann