JuliaRobotics/DistributedFactorGraphs.jl #133: FEATURE graph level small data

Affie, 2019-10-02:
What about adding graph level small data to all the DFG types?
eg in LightDFG:
```julia
mutable struct LightDFG{T <: AbstractParams, V <: AbstractDFGVariable, F <:AbstractDFGFactor} <: AbstractDFG
    g::FactorGraph{Int, V, F}
    description::String
    userId::String
    robotId::String
    sessionId::String
    addHistory::Vector{Symbol} #TODO: Discuss more - is this an audit trail?
    solverParams::T # Solver parameters
    **smallData::Dict{Symbol, String}**
end
```
Maybe a Dict{Symbol, String} for easier serialization?

GearsAD, 2019-10-07:
Example in #140 - it will help us address user/robot/session data too, which will be needed soon.

GearsAD, 2019-10-07:
TODO:
- MetaGraphDFG and SymbolDFG must implement
- FileDFG must serialize/deserialize
- CloudGraphsDFG must serialize/deserialize

Affie, 2020-02-07:
Partially implemented already.

closed 2020-05-01 by Affie