JuliaRobotics/DistributedFactorGraphs.jl #134: Standardize DFG behavior CRUD

Affie, 2019-10-03:
Decide on and standardize behavior on return and exceptions for CRUD

dehann, 2019-10-06:
Currently the behavior of GraphsDFG and CloudDFG is somewhat different.
We want it consistent, but i would not want the cloud breaking julia, but rather the ability to retry

dehann, 2019-10-06:
I like the cloud graphs addVariable in that it does not error, but not the automatic update if it exists.
I would propose the behavior is to @error on exists and return false.

dehann, 2019-10-06:
Update does not seem to change the graph structure in in-memory types which might be a bug?
It errors on GraphsDFG but creates with @warn on CloudDFG.
Create with @warn or @error seems logical

dehann, 2019-10-06:
Delete might just return nothing with @error agian?

dehann, 2019-10-06:
Hi Johan consistent definitely good, cloud @error with false a good start.  The rerun false might be under question at some point since initially addVariable used to return the variable object.  Maybe escalate the in memory to also be  @error.  Probably need a hard decision addVariable return type. Should limit to just one type I think.The retry can work well, although there is a discussion on whether user or driver has that responsibility.  If it’s small then maybe a keyword argument to disable retry?Not sure on the return type for delete, that could perhaps be nice if the DFGVariable is returned, or empty if failed?

dehann, 2019-10-06:
On retry i would say user responsibility, auto retry can maybe be implemented in RoME. What I meant was, rather return a false or nothing with an @error or @warn than throwing an exception that stops julia.

dehann, 2019-10-06:
Agree not to throw exception.  I’d suggest that that there is only one return type from the function for performance, but somehow indicates success or failure.

dehann, 2019-10-06:
On addVariable!: IIF returns the variable, DFG true/false. What is preferred. I would think if the function name is the same it should have the same behavior.

dehann, 2019-10-06:
definitely only one return type, but a Nothing Union eg. Union{Nothing, DFGVariable} might also be acceptable as the performance is not impacted that much since julia 0.7

dehann, 2019-10-06:
Right, union types thing... theres something I’m not sure about yet. There is a macro type_warn which shows type instability which is slow.  Do return Union in functions skip dynamic type inference — will be great if that’s the case.  Sounds like 0.7 was able to do that..?  At some point I’ll try the type_warn thing to see if Union return is stable, having a hard time understanding how that would be the case though.  As long as things are type stable I think it would be good?

dehann, 2019-10-06:
Its used in julia’s code base, eg findfirst.  I kind of remember reading that small type unions were now fast, but that might have been in containers.

dehann, 2019-10-06:
Found it: https://julialang.org/blog/2018/06/missing#fnref:splitting

dehann, 2019-10-06:
Oh cool, okay thanks.  Does this not also solve the dense adjacency matrix problem — ie Should use Union with isbits types like Missing.  Not to replace sparse.  I mean when going between sparse and dense representations.It does look like references are bad. So a Union{Missing, DFGVariable} would still be bad?  But a Union{Missing, Bool} would be good...  if I read that right.  It’s immutable types that the compiler can optimize.

dehann, 2019-10-06:
looks like Union{Missing, T} and Union{Nothing, T} would then be optimized in the same way? Preference as to which type? At the moment we're using a lot of Union{Nothing, T}

dehann, 2019-10-06:
General design pattern should be exception throwing - standard across languages - if we use the return checking approach we force the user to do the work. I get that Julia isn't good with exception handling yet (for some reason, hasn't stuck in Julia) , so let's go with the return code approach. Sound like a plan?

dehann, 2019-10-06:
just a consideration: as we introduce functions that talk to databases etc., we will end up trapping the potential exceptions in the addVariable function, and returning true/false, which may make it more difficult for users to debug why it failed (does the variable already exists? is it a comms issue? etc.)

dehann, 2019-10-06:
E.g. using the return true/false approach. If this works we can make the driver changes:  https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/136

dehann, 2019-10-06:
Hi, I think the point of that blog is Union nothing is bad but Union Missing is good.  Not 100% sure but at least Union Missing is good.I think try catch exception handling is good. Julia does the job, don’t think there is an issue with that.  My previous hold up with try statements was precisely the same issue of a graceful failure or one sentence error message but without reporting the call stack.  I’d prefer to use try catch but then just log the stack trace and message properly so we can debug more easily.  Try catch should not be used for duplicate adds or delete a missing value, right?So sounds like 2 questions then, if I have this right and feel free to disagree:
1. What is return types for graph cruds — considering missings?  Suggestion is Union{Missing, <: isbits}...
2. Missing is not an exception error but other errors should be caught internally vi try catch with full stack logs?  Currently only some of the DB code has this style...

Affie, 2019-10-07:
My question on IIF/RoME got missed, so repeating: On addVariable!: IIF returns the variable, DFG true/false. What is preferred. I would think if the function name is the same it should have the same behavior. I currently use the IIF/RoME variable that is returned. 
The options for addVariable!:
- Inconstant behavior; IIF return variable, DFG true/false or true exception.
- return true/false, use `addVariable!` then `getVariable!` -> I think this will have a big impact downstream.
- return variable everywhere, but question becomes how to deal with failure.

Affie, 2019-10-07:
My understanding of the blog is two parts:
1. functions: any 'small unions' are acceptable
2. arrays: only bit types.

returning nothing will automatically break the users code if a dfg variable is expected.

perhaps a flag that either throws exception or @error with nothing return?

dehann, 2019-10-07:
https://docs.julialang.org/en/v1/manual/style-guide/index.html#Don't-overuse-try-catch-1

dehann, 2019-10-07:
My question on IIF/RoME got missed, so repeating: On addVariable!: IIF returns the variable, DFG true/false. What is preferred. I would think if the function name is the same it should have the same behavior. I currently use the IIF/RoME variable that is returned.
The options for addVariable!:-Inconstant behavior; IIF return variable, DFG true/false or true exception.
-return true/false, use addVariable! then getVariable! -> I think this will have a big impact downstream.
-return variable everywhere, but question becomes how to deal with failure.

dehann, 2019-10-07:
HI Please continue this discussion in the issue thread.  Question did not get missed sorry -- answer is behaviour should be the same and we now need to make a design decision.

dehann, 2019-10-07:
I am personally not a big fan of returning union types yet, but chewing on it.

dehann, 2019-10-07:
Hi @GearsAD , do you have strong preference on return type for graph CRUD -- think we going to vote on this to resolve?  My feeling is we should return DFGVariable/DFGFactor type, but that leaves the issue of what to do when error occurs.  
- I don't know if Union{Missing, <:Node} is the the most performant and can create type instability, since it is not a 'small union'.
- Error handling inside the function is tangential issue, but there are two categories (do or dont do try catch inside CRUD):
  - Type 1: Functional syntactic errors
  - Type 2: 'not found' / 'duplicate / invalid type errors

## Alternative Suggestion for CRUD Return

How about always returning a `Tuple{Bool, <:DFGNode}`.  The bool flag states success and the second position is the relevant response.  ie
```julia
fg = initfg()
flag, var = addVariable!(fg, :x1, Pose2)
@assert flag
flag, var__ = addVariable!(fg, :x1, Pose2)
@assert !flag

# var__ should be empty container
```

## Only do try catch on CRUD Type 2 errors

Example DB, bad comms, etc.  Similar discussion for internal retries (#144)

Affie, 2019-10-07:
Summary of discussion with @GearsAD (Please feel free to update if I misunderstood something):
-`add...!` adds. 
-`update...!` updates, if it exists, adds (with @warn) if not. 
-`detete...!`
-`get...!`
All throws exception on fail or returns variable/factor on success. (to fit with RoME/IIF)

Affie, 2019-10-07:
small union: This optimization applies to all Unions of a small number of types, whether they are bits types or not. “Small” is defined by the MAX_UNION_SPLITTING constant, which is currently set to 4

dehann, 2019-10-07:
right, so i'm guessing `DFGVariable` or `DFGFactor` would exceed `MAX_UNION_SPLITTING`?

dehann, 2019-10-07:
> All throws exception on fail or returns variable/factor on success. (to fit with RoME/IIF)

RoME/IIF can change as needed.  Starting a new thread for that here #144

Affie, 2019-10-07:
My understanding Union{Nothing, Any} -> bad. Union{Nothing, DFGVariable} -> ok, spit in 2.

Affie, 2019-10-07:
A use case of interest:
```julia
var = addVariable!(localdfg, :x1, Pose2)
var = addVariable!(clouddfg, var)
```

dehann, 2019-10-07:
ah, I'm still stuck on the `isbits` aspect of `DFGVariable`.  The way I understand at the moment is that:

> arrays: only bit types.

means `Array{Union{<: isbits, <: isbits}}`.  

However, the following is still bad (because it involves pointers / references):
```julia
Union{Bool, Array{Bool}},
```
therefore the composite references type of `DFGVariable` will also be bad.  Thats my current understanding.  This is the images I keep getting:
```julia
f1(x)::Union{T1,T2} = ...

function f2(x)
  y = f1(x)
  doNext(y)
  println("thing")
end
```
still means that there is type instability on `y`.

dehann, 2019-10-07:
https://stackoverflow.com/questions/56424483/julia-code-warntype-issue-with-progressmeter

dehann, 2019-10-07:
Not sure what this means yet:
```juila
julia> f1(x)::Union{Bool, Vector{Float64}} = rand() > 0.5 ? rand(10) : false
f1 (generic function with 1 method)

julia> function f2()
         y = f1()
         @show y
         nothing
         end
f2 (generic function with 1 method)

julia> @warn_type f2()
ERROR: LoadError: UndefVarError: @warn_type not defined
in expression starting at REPL[5]:1

julia> @code_warntype f2()
Variables
  #self#::Core.Compiler.Const(f2, false)
  y::Union{}
  value::Union{}

Body::Union{}
1 ─     (y = Main.f1())
│       Core.Compiler.Const(:(value = y), false)
│       Core.Compiler.Const(:(y), false)
│       Core.Compiler.Const(:(Base.repr(%3)), false)
│       Core.Compiler.Const(:(Base.println("y = ", %4)), false)
│       Core.Compiler.Const(:(value), false)
└──     Core.Compiler.Const(:(return Main.nothing), false)
```

Affie, 2019-10-07:
This one shows in yellow for foo2 and foo3
```julia
struct Mytype
    a::Float64
    b::Vector{Float64}
end

function foo1(x::Float64)::Float64
    if x > 0
        return x
    else
        return 0
    end
end

function foo2(x::Float64)::Union{Nothing, Float64}
    if x > 0
        return x
    else
        return nothing
    end
end

function foo3(x::Float64)::Union{Nothing, Mytype}
    if x > 0
        return Mytype(x, randn(5))
    else
        return nothing
    end
end
```

dehann, 2019-10-07:
Also yellow:
```julia
function foo2(x::Float64)::Union{Missing, Int}
    if x > 0
       return 0
    else
       return missing
    end
end
```

dehann, 2019-10-07:
So does yellow mean be careful?  This would require all dispatches of downstream functions to also be defined, and you would not get "not found" but rather "function not defined" error when doing bad graph crud.

dehann, 2019-10-07:
@Affie , any strong feelings this way or that for the `Tuple` return option?

dehann, 2019-10-07:
I dont know what the tuple return will do for C++ cases when calling `.so`. Sure there is way to do it though.

Affie, 2019-10-07:
Small feeling against Tuple, as we have to check it then and it has a big impact downstream. Perhaps if it goes with a flag? (https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/134#issuecomment-539040678)

Affie, 2019-10-07:
The nothing return type was originally my idea for get variable:
```julia
var = getVariable!(localdfg, :x1)
var = addVariable!(clouddfg, var)
```
`addVariable!` expects a DFGVariable, therefore it will error in addVairable with the type instability not propagating. But as you say the error might be confusing, as you will get the @error one to the log and then a error exception stack trace. For robotics I like the idea of @error to a log file.

dehann, 2019-10-07:
Right, so that suggestion is always to always have `<:DFGNode` return type or exception on fail.  This would mean Type 2 errors are not caught inside the CRUD functions but rather given to the user.  I'm inclined to agree with that.

dehann, 2019-10-07:
I'm not so keen on changing the return type behavior with an optional flag, although have also done that in the past.  Mmmm, not sure.

GearsAD, 2019-10-08:
As per discussion:
- All functions will return either their type or Nothing (non critical failure)
- Update will add/update
- Add will fail if exists
- All functions will throw exceptions unless optional flag (noExceptions)==true

Sound good?

Affie, 2019-10-08:
If the nothing union is a problem. Before changing RoME/IIF to return tuple, I would suggest to return just a Bool. With getVariable perhaps returning a zero or uninitialized DFGVariable.    
Then the example would be: 
```juila
@assert addVariable!(localfg, :x1, :pose)
var = getVariable(fg, :x1)
addVariable!(cloudfg, var)
```
I still prefer union nothing for getVariable though.

Affie, 2019-10-08:
@GearsAD I would just ask that the flag is in the DFG structure itself. Either as a bool in the struct or in the new sessionData small data. What do you think? 
We can have something like: 
```julia
function isnoExceptions(dfg::AbstractDFG)::Bool
    haskey(getSessionData(dfg), :noException) ? true : false
end

#or 

isnoExceptions(dfg::AbstractDFG)::Bool =  dfg.noException
```

dehann, 2019-10-09:
@Affie, which level would you like to store this flag.  Sounds like a level 2 thing?

dehann, 2019-10-09:
I think `Union{Nothing, <:DFGNode}` is going to win.

dehann, 2019-10-09:
I agree with this:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/134#issuecomment-539343903

dehann, 2019-10-09:
@Affie , is this not perhaps a solver parameter rather than DFG issue: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/134#issuecomment-539378906
what I mean is:
```julia
getSolverParams(fg).noException = true
```

We can standardize different aspects such as `.dbg` later.

dehann, 2019-10-11:
so for ROS dev, if doing graph CRUD under a service methodology, hard failure of the service can be very annoying.  The best in that case would be to fail gracefully with an reply "NOT OK" ish.

https://github.com/JuliaRobotics/Caesar.jl/blob/feature/4Q19/rospoc/examples/ros/ServiceExample.jl#L19

Affie, 2025-09-02:
Error behaviour part of this was decided in #145 to be to throw typed errors. 
Return types are per verb and documented in #1085 

Custom errors are now thrown and return types per verb is also in progress. Closing this general issue and now case by case issues can be opened for any missed cases.