JuliaRobotics/DistributedFactorGraphs.jl #135: error: expected whitespace or a label

Affie, 2019-10-04:
```julia
ERROR: Any[Dict{String,Any}("message" => "Invalid input '0': expected whitespace or a label name (line 1, column 33 (offset: 32))\n\"match (node:User:Robot:0_test) where exists(node.label) and node.label = \"x0\" return id(node) \"\n                                 ^","code" => "Neo.ClientError.Statement.SyntaxError")]
```
For future reference, this error was caused by the session name starting with a number.

@GearsAD, should we maybe check for it in CloudGraphsDFG constructor?

GearsAD, 2019-10-20:
Oh, yeah, so there are pretty stringent requirements for userId, robotId, sessionId, and node label:
- Must start with a-z
- No spaces or special characters
- Can't be SESSION, ROBOT, USER

We should include as validation :+1:

dehann, 2019-11-17:
should allow `_`

dehann, 2019-11-21:
Just want to know if this was resolved for v0.5.0 or if its still upcoming? If anyone knows, please update the milestone on this issue (again).

closed 2020-05-03 by Affie