JuliaRobotics/DistributedFactorGraphs.jl #147: Designing PPE `VariableEstimate`

dehann, 2019-10-09:
Parametric Point Estimates (PPE) are stored in `VariableEstimate`, but there is a question of what data should be stored with it?  

Keeping in mind the data is stored in dictionary, example from current design:
```julia
 var.estimateDict[:default] = Dict{Symbol, VariableEstimate}(
    :max => VariableEstimate(:default, :max, varMax),
    :mean => VariableEstimate(:default, :mean, varMean),
    :modefit => VariableEstimate(:default, :modefit, varPpe))
```

Suggestions are:
- [ ] mean / max / modefit
- [x] actual value 
- [ ] softtypes which can point to (`:Pose2`) or contain manifolds (`Pose2().manifolds`)
- [ ] `solveKey`
...

dehann, 2019-10-09:
maybe a DFG v0.5.x thing

dehann, 2019-10-09:
devil's advocate -- how many flavors or variables would we store.  Skeleton / small data / PPE / big data ... ?  I'd like to apply some pressure towards stripping PPE down to values only

dehann, 2019-10-09:
FROM SLACK Sam:  Product of solve should go in estimate so we don’t need to deserialize it to use the result

Affie, 2019-10-09:
We definitely need softtype. We need to make sense of what the actual data is.

dehann, 2019-10-09:
what I mean is add softtype (with manifods) at this level:
https://github.com/JuliaRobotics/Caesar.jl/wiki/Data-Design-for-Caesar-RoME-IIF#level-1-should-be-quickly-searchable-over-thousands-of-nodes-possibly-returned-in-summary

Affie, 2019-10-09:
Softtype Missing in first class https://github.com/JuliaRobotics/Caesar.jl/wiki/Data-Design-for-Caesar-RoME-IIF#level-1-should-be-quickly-searchable-over-thousands-of-nodes-possibly-returned-in-summary

Affie, 2019-10-09:
I just listed it separately to make sure it is not only an array of numbers stored. As i can’t do anything with just the numbers and not any meaning behind them.

Affie, 2019-10-09:
What will be stored? Can it change that much? 
Can we not lump everything into one type and inherent from abstractsolverdata

dehann, 2019-10-09:
Why not just follow the 'data levels' we set out [to do in the beginning](https://github.com/JuliaRobotics/Caesar.jl/wiki/Data-Design-for-Caesar-RoME-IIF#list-of-important-data) and match that to the types of data the system returns

> how many flavors or variables would we store. Skeleton / small data / PPE / big data

- skeleton -- has no data,
- skeleton + PPE -- i level 1 and doesnt require any deserialization,
- small data -- is [level 2 requiring deserialization](https://github.com/JuliaRobotics/Caesar.jl/wiki/Data-Design-for-Caesar-RoME-IIF#level-2-still-in-graph-quickly-accessible-for-use-in-calculations),
- big data -- is level 3.

Affie, 2019-10-09:
Well, definitely needed with first class data. The summary data does not include the types. So i mistakenly though the type field was the softtype.

dehann, 2019-10-09:
I fully support the idea of PPE needs enough so you can visualize etc.  My question is  a little deeper, why not just use the 'data levels' because we want to do a lot more than just visualize...

dehann, 2019-10-09:
xref JuliaRobotics/RoME.jl#61

Affie, 2019-10-09:
I don’t know if you saw, there is a getSummarGraph that is supposed to be level 1.

dehann, 2019-10-09:
Ah haven't used that myself actually, thanks!

I updated the wiki to say we want softtype info at level 1?

Any feelings about the breakdown in [this comment](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/147#issuecomment-540033671)?

dehann, 2019-10-09:
whats the relationship of `DFGVariableSummary` vs VariableNodeData vs Skeleton and the 'data levels' ... which of the following are true?
- [ ] Skeleton -- no container yet
- [ ] Level 1 -- `DFGVariableSummary`
- [ ] Level 2 -- `VariableNodeData`
- [ ] Level 3 -- Big data TBD

Affie, 2019-10-09:
I have a draft skeleton in branch: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/1e11bc05711901850b44b2198c9c7b1e3e799197

Affie, 2019-10-09:
So far it will only work on LightDFG.

dehann, 2019-10-09:
lol, should we use (and add to wiki)?
alias skeleton = Level 0

Affie, 2019-10-09:
I was having a hard time deciding if there really exist such a thing as a skeletonDFGVariable. I added tags to it to make it more compatible. The factor only stores the order. I think you must say what you might need to do with it and what data is needed.

dehann, 2019-10-09:
> add tags

?

FYI, I updated [the wiki](https://github.com/JuliaRobotics/Caesar.jl/wiki/Data-Design-for-Caesar-RoME-IIF)

Affie, 2019-10-09:
This is what I meant by lump together:
```julia
struct VariablePPE <: AbstractVariableEstimate
    solverKey::Symbol
    softtype::Symbol
    max::Vector{Float64}
    mean::Vector{Float64}
    modemean::Vector{Float64}
    lastUpdatedTimestamp::DateTime
end
```

dehann, 2019-10-09:
we currently have `softtype` at level 2 (as introduced for computation.  Start by 
- [ ] duplicating it as `::Symbol` to Level 1? 
- [ ] if `::Symbol` we dont have manifold info, but thats okay -- we can add that later if it becomes really necessary.  Keep it smaller.  Visualizer can locally reconstruct manifolds if needed from the softtype symbol.

Affie, 2019-10-09:
>add tags

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/1e11bc05711901850b44b2198c9c7b1e3e799197/src/SkeletonDFGDev.jl#L13

dehann, 2019-10-09:
just an early FYI, `modemean` will likely get more structure too it as we start expanding that.  Can't really say what it will look like just yet.

Affie, 2019-10-09:
> modemean/modefit will likely get more structure
I figured, though to just repeat the data in the same array, say 1:4 is mode 1, 5:8 mode 2 etc

dehann, 2019-10-09:
Just remind me again what we decided there please: label is `:x1`, and tags are `[:LANDMARK; :VARIABLE; ...]` right?

Affie, 2019-10-09:
> duplicating softtype as ::Symbol to Level 1

yes that is what is needed as a minimum.

dehann, 2019-10-09:
> I figured, though to just repeat the data in the same array, say 1:4 is mode 1, 5:8 mode 2 etc

okay, can refine that later.

dehann, 2019-10-09:
So next part is:
- [ ] ` var.estimateDict[:default] = Dict{Symbol, VariableEstimate}(` vs. 
- [ ] hard structure `struct VariablePPE`

Personally I'm inclined to go for `Dict`, because as you have `struct VariablePPE`, you are again repeating Level 1 data, and there is no flexibility...

dehann, 2019-10-09:
im guessing you mean:
```julia
var.estimateDict[:default] = VariablePPE(....)
```

Affie, 2019-10-09:
> no flexibility
That is why I asked how often will it change, because flexibility is slightly harder. 
Flexibility comes from <: AbstractVariableEstimate.

dehann, 2019-10-09:
The `modefit` part will change a few times.  `max` and `mean` are pretty stable i think.  I'm concerned we start doing other things in the realm of PPE, but not just `mean/max/modefit`.

Ex, cc @pvazteixeira had spoken about going into set-theoretic probability and confidences stuff.

Affie, 2019-10-09:
As a user learning to use the software, the flexibility of a Dict is sometimes hard. You don't know what fields there are and exactly what they mean. 
If max mean is stable then maybe a VariableEstimateMaxMean <: AbstractVariableEstimate

dehann, 2019-10-09:
i think we need a link to examples here if possible please Pedro, but the point is PPE is likely to change a few times in the short/medium term.

dehann, 2019-10-09:
> user learning to use the software, the flexibility of a Dict is sometimes hard

understood.  Perhaps something like this?
struct VariablePPE <: AbstractVariableEstimate
    lastUpdatedTimestamp::DateTime
    max::Vector{Float64}
    mean::Vector{Float64}
    modefit::Vector{Float64}
    other::Dict{Symbol, Vector{Float64}}
end

dehann, 2019-10-09:
haha is `lastUpdatedTimestamp::DateTime` not just Level 1 data too?

Affie, 2019-10-09:
Maybe we must first decide if data is repeated? I think @GearsAD would like it repeated for serialization? I tend to lean to not repeating stuff.

Affie, 2019-10-09:
>haha is lastUpdatedTimestamp::DateTime not just Level 1 data too?

No, that is when the calculation was updated as I understand.

dehann, 2019-10-09:
+1 not repeating and putting it in Level 1 (which you should have available when getting PPE data anyway ... i think / hope?)

dehann, 2019-10-09:
Another thing to remember about `Dict` vs hard `struct VariablePPE`, is that only Julia understands `VariablePPE` -- which in turn immediately means serialization.  Does that not mean we have to use Dict in all cases here for Level 1 requirements.  PPE must be Level 1 by design.

dehann, 2019-10-09:
> that is when the calculation was updated as I understand.

Agreed, but does that not apply to all Level 1/2/3

Affie, 2019-10-09:
I don't think so, it looks like it serializes to a dict fieldname => value.

Affie, 2019-10-09:
> Agreed, but does that not apply to all Level 1/2/3?

What do you mean?

dehann, 2019-10-09:
I mean lastupdated timestamp

Affie, 2019-10-09:
The timestamp I wanted in level one was the actual timestamp of the pose. It should be fixed when created by the robot and not allowed to change.

dehann, 2019-10-09:
Sure, I'm asking if we should not add a second `lastupdated` timestamp to Level 1?

dehann, 2019-10-09:
search everything not updated in 10 days and recalculate -- kinda ideas...

Affie, 2019-10-09:
Ahh, what about big data? small data? all can be updated at different times. Also :default and other solver parameters

dehann, 2019-10-09:
oh that is true, did not think of that thanks...

dehann, 2019-10-09:
sure leave `lastupdated` inside PPE then.

Affie, 2019-10-09:
So what about the data type?

Affie, 2019-10-09:
Just for reference: 
```julia
mutable struct DFGVariableSummary <: AbstractDFGVariable
    label::Symbol
    timestamp::DateTime
    tags::Vector{Symbol}
    estimateDict::Dict{Symbol, Dict{Symbol, VariableEstimate}}
    _internalId::Int64
end
```

dehann, 2019-10-09:
I'd like to hear @GearsAD 's opinion on whether we can search over something like `VariablePPE` in Neo4j.  I'd say a critical requirement of Level 1 data is that JSON packing / unpacking is the most complicated the data can be, barring data in Level 2 and Level 3 that might appear.

dehann, 2019-10-09:
yes, question extends to definition of what is Level 1 -- if the use case is we want to search over it efficiently without deserializing:
```julia
DFGVariableSummary <: AbstractDFGVariable
```
might be contrary in that case.

dehann, 2019-10-09:
understand there is a short term vs long term design driver here.

Affie, 2019-10-09:
what about the `VariableEstimate` in `estimateDict::Dict{Symbol, Dict{Symbol, VariableEstimate}}`

dehann, 2019-10-09:
yes, that has the same serialization question -- Sam had put that in there, so sure he is happy with it, just want to hear his opinion on the [data type decision](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/147#issuecomment-540072334) with don't de-serialize data in Level 1 concept.

Affie, 2019-10-09:
> yes, question extends to definition of what is Level 1 -- if the use case is we want to search over it efficiently without deserializing:
> 
> ```julia
> DFGVariableSummary <: AbstractDFGVariable
> ```
> 
> might be contrary in that case.

Sorry I don't understand.

dehann, 2019-10-09:
I think the JSON marshalling and unmarshalling is powerful enough to unpack these basic types back and forth to Dict.  In order for something like Neo4j to search over Level 1 data, it needs to be JSON'able -- i.e. Dict.  By choosing Dict data type you are kinda ensuring that it is pretty close to JSON.  By introducing types like `DFGVariableSummary /  VariableEstimate / VariablePPE` it might create problems with JSON, and hence Neo4j -- although i don't know enough but @GearsAD will have the answer.  Hence request to hear opinion before we decide on this or that way.

My vote is 
- on data type either or, as long as there is at minimum the `other::Dict{Symbol, Vector{Float64}}` option in the hard type case.
- on `lastupdated`, yes keep it within PPE,
- on `softtype`, yes introduce as duplicate as repeat data from Level 2 but only as name Symbol in Level 1.
- on storing PPE, do that with top level `Dict`:
  - `var.estimateDict[:default] = <<DATA TYPE OPTION>>`

Affie, 2019-10-09:
Note `DFGVariableSummary` is the level 1 data structure currently in use. I just added it here for reference.

[50 more replies]

PR #183 (2019-10-26): Closes #147 PointParametricEst implementation
Closes #147 
Closes #158 replace pr with new design
Closes #148 replace pr with new design

**Note**
Just the struct so far, TODO integrate, testing, compare, etc.
commit 83274f28 (2019-10-26): Merge pull request #183 from JuliaRobotics/feature/147_PointParametricEst

Closes #147 PointParametricEst implementation
closed 2019-10-26 by GearsAD