JuliaRobotics/DistributedFactorGraphs.jl #149: DFGVariable softtype field has abstract type

Affie, 2019-10-10:
Should DFG not export an abstract type with IIF InferenceType inheriting from it? I don't like that it does not have a type. see https://docs.julialang.org/en/v1/manual/performance-tips/#Avoid-fields-with-abstract-type-1
eg.`AbstractSoftType` then in IIF `InferenceType <: AbstractSoftType`
even `VariableNodeData{T<:Any}` will be better I think.

I just realized that kind of makes the softtypename field redundant.

_Originally posted by @Affie in https://github.com/_render_node/MDI0OlB1bGxSZXF1ZXN0UmV2aWV3Q29tbWVudDMzMzQxNDkyMQ==/comments/review_comment_

Affie, 2019-10-10:
It might be nice if  DFGVariable{T<:AbstractSoftType} also. Then the type data is inherently saved in the variable.

dehann, 2019-10-10:
Hi, yes the lack of type is no good for performance.  `DFGVariable{T<: ... }` is the rigth way to go.  Not sure if we need to introduce so many layers of abstraction.  Just inhereting from `InferenceVariable` might be good enough.  Thanks for bringing this up!

Affie, 2019-10-10:
This is why it is blank I thinik, InferenceVariable is in IIF. Thats why the extra layer.

dehann, 2019-10-10:
`InferenceVariable` will likely move to DFG if it hasn't already happened.

dehann, 2019-10-10:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/9afd7a380cc5d124c74bc3a1a466202cd5370fe2/src/entities/DFGFactor.jl#L8

Affie, 2019-10-10:
Ok, then its not a problem. I didn't think to look in DFGFactor.jl at all.

dehann, 2019-10-10:
yeah, weird location, but probably some good reason somewhere.

Affie, 2019-10-10:
I'll quickly fix issue.

Affie, 2019-10-10:
Unfortunately, It looks like it will be a breaking change.

Affie, 2019-10-10:
Unfortunately it would be a breaking change:
```julia
function addVariable!(dfg::G,
                      lbl::Symbol,
                      softtype::T;
                      N::Int=100,
                      autoinit::Bool=true,  # does init need to be separate from ready? TODO
                      ready::Int=1,
                      dontmargin::Bool=false,
                      labels::Vector{Symbol}=Symbol[],
                      smalldata=Dict{String, String}(),
                      checkduplicates::Bool=true  )::DFGVariable where
                        {G <: AbstractDFG,
                         T <: InferenceVariable}
  #
  v = DFGVariable(lbl)
#needs to add the parameter
DFGVariable{T}(lbl)
```

Affie, 2019-10-10:
Is it ok if i made the milestone 0.5.0? It’s mostly performance and is likely to break a lot of code.

Affie, 2019-10-11:
see #149. I still think making the type parametric will be of advantage since it is used in the core algorithm.

dehann, 2019-10-13:
>  use DFGVariable(lbl, softype)

I'm not too hot on double template here.  Single `{T}` for variable softtype is okay.

>  the type parametric will be of advantage since it is used in the core algorithm

PPE is not used in the core algorithm, if thats what you are referring too.  PPE is for user convenience -- reduction from the marginal belief estimate.

Affie, 2019-10-13:
> I’m not too hot on double template here...

I don’t know how to do it then. Is the whole DFGVariable not of one type. 

DFGvariable has an abstract type softtype::Any. It is used everywhere not just for ppe (ppe does not have a softtype field). Or am i missing something?

Affie, 2019-10-14:
Quickly cleaned up and committed what I had so far, see commit: https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/dcd97da443e80d3aabe306ec187fc9c283424c2e

dehann, 2019-10-14:
> I still don't see the problem with <:InferenceVariable

Bad: `solverDataDict::Dict{Symbol, VariableNodeData{<:InferenceVariable}}`

#### Which is very different from templating:

Good: 
```julia
mutable struct VariableNodeData{T<:InferenceVariable}
  ...
  softtype::T
  ...
end
```

dehann, 2019-10-14:
with this:

> I'm not too hot on double template here

I meant: the PPE compat design issue is not solved by introducing templating.  We argued it out.  You would have to change the (**now old**) VariableEstimate again and again and again, which will require new minor revision number every time.  The only way to avoid that (which we could come up with) is to use a dict.  Hence PPE changed to PointParametricEstimate which now contains a Dict (thats PR 158).
- the performance of something like `MyType{T1 <: Abstract9, T2 <: Abstract3}` is very good;
- the performance of containers with abstract types is bad, `struct BadType    val::Abstract    end`

commit 2a0f0038 (2019-11-18): Merge pull request #184 from JuliaRobotics/feature/149_DFGVariableSofttype

Closes #149 DFGVariable softtype parametric type
closed 2019-11-18 by GearsAD