JuliaRobotics/DistributedFactorGraphs.jl #156: unpackFactor/unpackVariable need refinement

GearsAD, 2019-10-12:
- We pass incrementalinference in as a module to deserialize the factor... it's a little janky
- When deserializing the factors it uses the graph that's passed in to find the variables related to the factor. It's a little strange passing in a graph when deserializing.
- When deserializing a factor you need to call IncrementalInference.rebuildFactorMetadata!(dfgLoadInto, factor) to make it useable - this is the one that's going to catch everyone

dehann, 2019-10-12:
could be a IIF abstraction thing too.  WIll try fix in general sweep at some point?

GearsAD, 2019-10-13:
Yes, sounds like a plan. For now you need to pass IIF to unpackFactor - we may want to clean up that up in future.

Affie, 2020-05-21:
> - We pass incrementalinference in as a module to deserialize the factor... it's a little janky

Removed from `loadDFG!`

> - When deserializing the factors it uses the graph that's passed in to find the variables related to the factor. It's a little strange passing in a graph when deserializing.
 
`getTypeFromSerializationModule` does not use the graph any more
but still needed for `getFactorOperationalMemoryType(dfg)` to dispatch on for the convert functions.

> - When deserializing a factor you need to call IncrementalInference.rebuildFactorMetadata!(dfgLoadInto, factor) to make it useable - this is the one that's going to catch everyone

This is now dispatched on `AbstractSolverData` type #449

closed 2020-05-21 by Affie