JuliaRobotics/DistributedFactorGraphs.jl #166: updateVariableSolverData! behavior

Affie, 2019-10-18:
@GearsAD, From tests I see we may have different views on what  `updateVariableSolverData!` does.
I see update as only add to and not delete existing keys:
- If `var1` has only :default and you update it with `var2` that only has :second. `var1` should get a :second and :default should remain.
- If `var1` has only :default and you update it with `var2` that has :defualt. `var1` :default should be replaced/updated 

cc @dehann

GearsAD, 2019-10-19:
No, because in that case there is no way to delete. We've discussed this and an update is an atomic transaction that performs a complete update, shouldn't make it do unexpected things.

GearsAD, 2019-10-19:
I'm not sure why you want to make it smarter than a simple CRUD update, I can't see the benefit.

Affie, 2019-10-20:
I though the idea was to have different solves. So if one dfg runs solve1 and another solve2 what do you keep, solve1 or solve2? I thought both. If you just replace the whole dictionary it will update just one. So it is still a complete update but of one solve and not all solves. 
But let me know if i misunderstand the whole solves thing.

Affie, 2019-10-20:
> No, because in that case there is no way to delete

Then add a delete. Update should not delete, it should update.

GearsAD, 2019-10-20:
Okay - so the concern is that on datastructures CRUD behavior is standard, and we're deviating from how you generally treat a dictionary. It's fair that you want to merge, so how about this going to change the name and keep the behavior the same.

closed 2019-10-22 by GearsAD