JuliaRobotics/DistributedFactorGraphs.jl #169: document save / load DFG

dehann, 2019-10-18:
I needed this to work today, sigh...
```julia
help?> saveDFG
search: saveDFG savejld has_negative_edge_cycle

  No documentation found.

  DistributedFactorGraphs.saveDFG is a Function.

  # 1 method for generic function "saveDFG":
  [1] saveDFG(dfg::G, folder::String) where G<:AbstractDFG in DistributedFactorGraphs at /home/dehann/.julia/dev/DistributedFactorGraphs/src/FileDFG/services/FileDFG.jl:3
```

GearsAD, 2019-12-01:
Sorry it took so long, snuck it in with an update:
```
help?> saveDFG
search: saveDFG

  saveDFG(dfg, folder)
  

  Save a DFG to a folder. Will create/overwrite folder if it exists.

  Example
  ≡≡≡≡≡≡≡≡≡

  using DistributedFactorGraphs, IncrementalInference
  # Create a DFG - can make one directly, e.g. GraphsDFG{NoSolverParams}() or use IIF:
  dfg = initfg()
  # ... Add stuff to graph using either IIF or DFG:
  v1 = addVariable!(dfg, :a, ContinuousScalar, labels = [:POSE], solvable=0)
  # Now save it:
  saveDFG(dfg, "/tmp/saveDFG")
```

GearsAD, 2019-12-01:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/228/files#diff-f898c9a7e4c880c61f98ab8d917ee42aL32-R63

closed 2019-12-01 by GearsAD