JuliaRobotics/DistributedFactorGraphs.jl #172: Docs failing due issues in Light/Symbol/etc DFG

dehann, 2019-10-21:
Hi, I'm trying to build docs in Caesar package and came across this.  Any quick ideas, or should we pick through them all and fix one by one?

cc @Affie might know best on this one?

```julia
┌ Error: doctest failure in ~/.julia/dev/DistributedFactorGraphs/src/SymbolDFG/SymbolFactorGraphs/SymbolFactorGraphs.jl:249-261
│ 
│ ```jldoctest
│ julia> using LightGraphs
│ 
│ julia> g = SimpleGraph(2);
│ 
│ julia> add_edge!(g, 1, 2);
│ 
│ julia> rem_edge!(g, 1, 2)
│ true
│ 
│ julia> rem_edge!(g, 1, 2)
│ false
│ ```
│ 
│ Subexpression:
│ 
│ using LightGraphs
│ 
│ Evaluated output:
│ 
│ ERROR: ArgumentError: Package LightGraphs not found in current path:
│ - Run `import Pkg; Pkg.add("LightGraphs")` to install the LightGraphs package.
│ 
│ Stacktrace:
│  [1] require(::Module, ::Symbol) at ./loading.jl:876
│ 
│ Expected output:
│ 
│ 
│ 
│   diff =
│    ERROR: ArgumentError: Package LightGraphs not found in current path:
│    - Run `import Pkg; Pkg.add("LightGraphs")` to install the LightGraphs package.
│    
│    Stacktrace:
│     [1] require(::Module, ::Symbol) at ./loading.jl:876
└ @ Documenter.DocTests ~/.julia/packages/Documenter/IGqbY/src/DocTests.jl:364
┌ Error: doctest failure in ~/.julia/dev/DistributedFactorGraphs/src/SymbolDFG/SymbolFactorGraphs/SymbolFactorGraphs.jl:249-261
│ 
│ ```jldoctest
│ julia> using LightGraphs
│ 
│ julia> g = SimpleGraph(2);
│ 
│ julia> add_edge!(g, 1, 2);
│ 
│ julia> rem_edge!(g, 1, 2)
│ true
│ 
│ julia> rem_edge!(g, 1, 2)
│ false
│ ```
│ 
│ Subexpression:
│ 
│ g = SimpleGraph(2);
│ 
│ Evaluated output:
│ 
│ ERROR: UndefVarError: SimpleGraph not defined
│ Stacktrace:
│  [1] top-level scope at /home/dehann/.julia/dev/Caesar/docs/make.jl:15
│ 
[... 362 lines cut ...]
│ 
│ julia> rem_vertex!(g, 2)
│ false
│ ```
│ 
│ Subexpression:
│ 
│ rem_vertex!(g, 2)
│ 
│ Evaluated output:
│ 
│ ERROR: UndefVarError: rem_vertex! not defined
│ Stacktrace:
│  [1] top-level scope at /home/dehann/.julia/dev/Caesar/docs/make.jl:15
│ 
│ Expected output:
│ 
│ false
│ 
│   diff =
│    falseERROR: UndefVarError: rem_vertex! not defined
│    Stacktrace:
│     [1] top-level scope at /home/dehann/.julia/dev/Caesar/docs/make.jl:15
└ @ Documenter.DocTests ~/.julia/packages/Documenter/IGqbY/src/DocTests.jl:364
```

Affie, 2019-10-22:
Sorry @dehann, thats my bad. I didn't want to redo the docs from scratch so I copied and never got around to updating it.
The thing is SymbolDFG is not expoted  and I think the docs should not be build. 
Can it be skipped in Caesar or should we do a patch release where we can remove all of them.
I think Sam's idea is to completely remove the SymbolDFG and MeteGraphsDFG code.

dehann, 2019-10-23:
Oh, dont worry about Caesar docs in the short term.  I have a workaround in place.  Doc failures are ignored by Travis, but passing well enough to actually deploy updates to Caesar docs.  As long as there is some medium term plan here to fix, then lets go with most sensible and minimum energy.

dehann, 2019-10-23:
does this have anything to do with JuliaRobotics/IncrementalInference.jl#412 ?

dehann, 2019-11-21:
if interested, Caesar docs are currently failing because of a unsolvable JSON.jl issue.  Two dependencies want either JSON v0.20.0 or v0.21.0 -- so dont think this issue here is a problem with Caesar docs at this time.

Affie, 2020-02-07:
I think all of them were removed, this should be resoled in v0.6.0

PR #291 (2020-02-12): Development updates for DFG v0.6
partial fix for #241 
fix #239
close #229
implement #211
fix #172
part of #234
closed 2020-02-12 by Affie