JuliaRobotics/DistributedFactorGraphs.jl #178: Release 0.4.2

GearsAD, 2019-10-22:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2019-10-22:
Registration pull request created: [JuliaRegistries/General/4630](https://github.com/JuliaRegistries/General/pull/4630)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.2 -m "<description of version>" 9fd1b28ad853e78e58be3814bfb4f1872bfe24f1
git push origin v0.4.2
```

dehann, 2019-10-22:
Oops, https://github.com/JuliaRegistries/General/pull/4630#issuecomment-545200849

> The following dependencies do not have a compat entry that has an upper bound: Colors, GraphPlot, > JSON2, LightGraphs

dehann, 2019-10-22:
Once you have it, you can just recall registrator here (just as done above https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/178#issue-510986807)

GearsAD, 2019-10-23:
@JuliaRegistrator register branch=release/v0.4

JuliaRegistrator, 2019-10-23:
Registration pull request updated: [JuliaRegistries/General/4630](https://github.com/JuliaRegistries/General/pull/4630)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.2 -m "<description of version>" afa0634751f0c937add22c4d856e16089e134966
git push origin v0.4.2
```

dehann, 2019-10-23:
Yeah, the comment does permanently require human intervention.  So future my plan is to not make mistakes, but if they occur to quickly fix and then refute the registration.  Might be useful if Zepto gives a time of automerge delay in the various cases.

closed 2019-11-17 by GearsAD