JuliaRobotics/DistributedFactorGraphs.jl #185: Searching estimates: PPE Database design dicussion

GearsAD, 2019-10-29:
Discussion on call. It's about time we optimize our database structures - we need to look at how we search/retrieve data.

Ideally if we're looking at temporal+spatial retrievals, we need to start indexing against that. Think we're all in agreement - we have PointParametricEstimates, how do we save them to make them searchable?

From the searchable side, we effectively want to build indices in Neo4j, like this:
```
create index on :Variable(Estimate)
```

Given that we're currently packing estimates into JSON blobs, that won't really work. @Affie suggested moving the estimate to a new PPE node, and relating that to each variable. Looking at that in more detail:
- If we do that we can index the PPE's independently
- We can explode the estimate vector and index against it
- We can make more specialized indexes (think spatial indices using the Point type) - https://neo4j.com/docs/cypher-manual/current/schema/index/#schema-index-spatial-distance-searches-single-property-index

E.g.:
```
{
  "X": 5,
  "Y": 1,
  "Z": 7,
  "type": "Mean",
  "cart": point({srid:9157, x:5, y:1, z:7}),
  "key": "DEFAULT"
}
```

We can make indices against X,Y,Z, type, cart, and key, which would make searching very efficient.

So, that means I'd suggest we move forward changing the code to persist a PPE node. I'd like that we start thinking about how we can specify (with macro's) what in the estimate could be wrapped into a 2 or 3-space coordinate? 

Things to discuss:
- [ ] Are we all ok making PPE's nodes?
- [ ] Are we ok exploding the estimates so we can search them? 
- [ ] Discussion: Can we make macros to make estimate types possibly wrap to a cartesian Point type in Neo4j, so we can make the searching a lot quicker?

Affie, 2019-10-30:
Related: what do we do with the SolverDataDict  (and possibly BigDataDict)?
- It does not have to be indexed, not lv 1.
- It can therefore be data blobs such as JSON (like it currently is).
- However, we might want to have quick access to the `solverkeys` (:default, :second)

Is `ready` and `backendset` not per 'solve'?

dehann, 2020-03-03:
related to #215

Affie, 2020-08-02:
This is implemented and cleaned up in #517 with solver data also added.

closed 2020-08-02 by Affie