JuliaRobotics/DistributedFactorGraphs.jl #186: [FEATURE] sort variables/factors by time-added

dehann, 2019-11-06:
BREADCRUM to remember:  I frequently run into the situation where it would be super useful if I could just sort the variables or factors in a list according the timestamp (or FIFO queue) they were added to a specific graph.

Consider an intermediate string of poses and odometry factors, and imagine trying to recover the earliest 4 or last 5...
```julia
poses = Symbol[Symbol("x$i") for i in 10:20]
odos = intersect(map(s->ls(fg,s), ls(fg, Pose2Pose2))
sortByTime!(odos)

# first 4 odos
odos[1:4]

# last 5
odos[(end-5):end]
```

pvazteixeira, 2020-02-03:
I just ran into this issue and this is how I addressed it:

```julia
import Base.isless

function isless(a::DFGVariable, b::DFGVariable)
    return isless(a.timestamp,b.timestamp)
end
```

Now you can do stuff like `sort(allVariables)`

Affie, 2020-02-03:
Maybe consider using the `lt` keyword argument over overwrite isless, see https://docs.julialang.org/en/v1/base/sort/#Base.sort!
Then we can have natural sort on labels, timestamp sort, sort by position, etc. We just need to sort out the api.

pvazteixeira, 2020-02-03:
Good point; I'll take a closer look. In any case, I'd argue for sorting by timestamp as the default.

Affie, 2020-02-07:
If you have any ideas on API, I can go ahead and add this to v0.6.0?

I think just exposing a few `isless` functions that will work with `sort` with documentation should do the trick:
```julia
- sort(allVariables, lt=ls_timestamp)
- sort(allVariables, lt=ls_label)
```
Or do we want the noun as in `sortVariables`, 
I'm against unnecessary long names such as `sortVariableByTimestamp`

@dehann, how does this fit in with the API standardization. 
I'm in favor of sticking to base as much as possible and not having extra functions to maintain.

pvazteixeira, 2020-02-07:
Having at least those two (timestamp and label) would be useful - maybe default to timestamp so that a simple `sort(allVariables)` works - or are you explicitly trying to avoid that?

dehann, 2020-02-12:
hi sorry for being late to the party here, have just been using sortDFG since Base.sort is useful in other cases.  Not 100% match to verbNoun definitions but close enough?

dehann, 2020-02-12:
i personally like sort and sortDFG as separate functions since both can be called on vector symbol

Affie, 2020-02-12:
do we have sortDFG default to sort by timestamp then? I'm hesitant to overwrite `isless` for Variables since `isless` is not cleanly defined

Affie, 2020-02-12:
What do you want to do with `sortDFG` that sort cannot do?

Affie, 2020-02-12:
something like:
```julia
sortDFG(v::Vector{Symbol}) = sort(v, lt=DistributedFactorGraphs.natural_lt) 
sortDFG(v::Vector{DFGVariable}) = sort(v, by=getTimestamp)
```
so just a wrapper?

dehann, 2020-02-12:
i a little worried about overriding Base.sort?

dehann, 2020-02-12:
just a wrapper is fine, based on what @Affie already did here:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/master/test/testBlocks.jl#L768-L777

dehann, 2020-02-12:
What is default sorting method (based on @pvazteixeira 's question)

Lets vote:
- Timestamps: Dehann, Pedro
- `.label`

Johan is with the masses, but strongly recommends we DO NOT override `Base.isless`.

Decision is we use Timestamp as default ordering and can change to label::Symbol using a keyword argument

dehann, 2020-02-12:
Related to #241

commit 294c1f3d (2020-02-13): Merge pull request #295 from JuliaRobotics/feature/20Q1/sort#186

sortDFG wrapper close #186 and close #241
commit 73025343 (2020-02-13): sortDFG wrapper close #186 and close #241
closed 2020-02-13 by Affie