JuliaRobotics/DistributedFactorGraphs.jl #195: Release 0.5.0

GearsAD, 2019-11-18:
@JuliaRegistrator register branch=release/v0.5

JuliaRegistrator, 2019-11-18:
Registration pull request created: [JuliaRegistries/General/5522](https://github.com/JuliaRegistries/General/pull/5522)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.5.0 -m "<description of version>" d1abeb25331c520b4ccf35a337e007691f30d95c
git push origin v0.5.0
```

GearsAD, 2019-11-18:
@JuliaRegistrator register branch=release/v0.5

JuliaRegistrator, 2019-11-18:
Registration pull request updated: [JuliaRegistries/General/5522](https://github.com/JuliaRegistries/General/pull/5522)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if [Julia TagBot](https://github.com/apps/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.5.0 -m "<description of version>" 44c403b3b9b441214c67242aff7472e9ed30f4fc
git push origin v0.5.0
```

closed 2019-11-21 by dehann