JuliaRobotics/DistributedFactorGraphs.jl #201: rename ready and backendset to solvable and solveinprogress

dehann, 2019-11-19:
Also:
- [ ] rename `ready` --> `solvable` in DFGVariable,
- [ ] rename `backendset` --> `solveinprogress`,
- [ ] move `solveinprogress` down one layer to `VariableNodeData`

cc @Affie

dehann, 2019-11-19:
xref JuliaRobotics/IncrementalInference.jl#438

dehann, 2019-11-22:
@GearsAD says there is a problem with putting solveinProgress in VND -- cant be searched for in Neo4j i think?

GearsAD, 2019-11-22:
This is good. If we move solveinprogress then when you call getneighbors with a  filter, we need to specify what solvekey to use to check if it both exists and is set. E.g. for this call:
```
function getNeighbors(dfg::GraphsDFG, node::T; solvable::Union{Nothing, Int}=nothing, backendset::Union{Nothing, Int}=nothing)::Vector{Symbol}  where T <: DFGNode
```

dehann, 2019-11-22:
- so there is no problem moving `solveInProgress` down to VND?
- do we need to be able to search for it in Neo4j?

GearsAD, 2019-11-22:
Not just in Neo4j, in general. I'm quickly looking over it but something like this would be needed if we want to then search by `solveinprogress`. For example, I think we would need to change the functions to look like this:
```
"""
    $(SIGNATURES)
Retrieve a list of labels of the immediate neighbors around a given variable or factor.
"""
function getNeighbors(dfg::GraphsDFG, node::T; solvable::Union{Nothing, Int}=nothing, solveinprogress::Union{Nothing, Int}=nothing, solvekey::Union{Nothing, Symbol}=:default)::Vector{Symbol}  where T <: DFGNode
    if !haskey(dfg.labelDict, node.label)
        error("Variable/factor with label '$(node.label)' does not exist in the factor graph")
    end
    vert = dfg.g.vertices[dfg.labelDict[node.label]]
    neighbors = in_neighbors(vert, dfg.g) #Don't use out_neighbors! It enforces directiveness even if we don't want it
    # Additional filtering
    neighbors = solvable != nothing ? filter(v -> v.dfgNode.solvable == solvable, neighbors) : neighbors
    neighbors = solveinprogress != nothing ? filter(v -> haskey(v.dfgNode.solverDataDict, solvekey) && v.dfgNode[solveKey].solveinprogress == solveinprogress, neighbors) : neighbors
    # Variable sorting (order is important)
    if node isa DFGFactor
        order = intersect(node._variableOrderSymbols, map(v->v.dfgNode.label, neighbors))
        return order
    end

    return map(n -> n.dfgNode.label, neighbors)
end
```

(May not work, just example for discussion)

dehann, 2019-11-22:
Separate performance issue:
> `solvable != nothing ? filter(v -> v.dfgNode.solvable == solvable`
we should just make sure that these cases do not result in type instability.  Check with `@warn_type` (? or near abouts)

dehann, 2019-11-22:
> For example, I think we would need to change the functions to look like this:

i think that is fine.

> Not just in Neo4j

in which cases do we need this actually to work in Neo4j?  Perhaps not even a real requirement.  The history was that `fullLocalGraphCopy` used `ready` and `backendset` to define a intersect of the graph that needed to be copied.  With new DFG and CSM I dont think that is a requirement any more...

GearsAD, 2019-11-22:
Unless you need it, I don't think so. It was just older requirement. I'd say we change up getNeighbors to use the new signature.

dehann, 2019-11-22:
okay, and how do you feel about the type instability thing?

Affie, 2019-11-22:
What is the use case of filtering with `solveinprogress` in `getNeighbors`?

GearsAD, 2019-11-23:
@dehann - I'll check for the type instability thing in that, think its fine but now I'm curious.

@Affie  - It grew out of something in the past. However, @dehann, would that be a valid use-case now? If not, removing that filtering would be great.

GearsAD, 2019-11-23:
Okay, while we're here. @dehann 

- Using solvable in this way means that it's a lower bound when filtering - filtering on solvable=0 means that you'll get both solvable=1 and solvable=0. That is correct?
- Is there any use for getNeighbors with solvable and solveInProgress? If not, I'd like to remove it
- ls/getVariables/getFactors - should this have a solvable filter for general use?
- Can we safely remove MetaDFG and SymbolDFG?

Affie, 2019-11-23:
Hi, MetaDFG I would say yes. We might still need SymbolDFG in the future, but we can put it back then. So yes, remove.

GearsAD, 2019-11-24:
Was reading https://julialang.org/blog/2018/08/union-splitting

My understanding is that Union{Int, Nothing} is therefore ok. Thoughts?

dehann, 2019-11-25:
we can probably close this issue with DFG v0.5.1 right?

GearsAD, 2019-11-26:
Yep. Fixed in 0.5.1

closed 2019-11-26 by GearsAD

dehann, 2019-12-04:
hi @GearsAD , i reread and see one question went unanswered.  We definitely still need `solveinprogress` and I understand that this is now inside each solveKey VND in the dict -- which is fine/good.

GearsAD, 2019-12-18:
Yep, discussed it a bit :+1: