JuliaRobotics/DistributedFactorGraphs.jl #202: Should :default => VariableNodeData be populated in DFGVariable constructor

Affie, 2019-11-21:
referring to `:default => VariableNodeData{T}()`
I would think that creating an empty dictionary is more appropriate since IIF will populate the values regardless. I'm experimenting with ways to implement other algorithms and turning `:default` off completely. 
I just want to know if there was a good reason for populating it.

dehann, 2019-11-21:
I don't have any meaningful suggestions on this, so either way is good with me.

dehann, 2019-11-21:
The only reason I can see for populating it is to force a type T.  As long there is not an anonymous type instability later when the Dict is populated it should be good.

dehann, 2019-11-21:
initially populating the Dict with an abstract would also be bad

Affie, 2019-11-22:
About your comment from slack:

> in general yes its bad to have solverDataDict contain different softtypes -- although in this case I think that will be rare.  Only if we use different variable softtypes in different supersolves.  My suggestion is just ot raise a warning at create time if that occurs and then allow it to be slow.  All other cases should have solverDataDict  contain all the same softtype -- e.g. Pose2, Point2 ...

Do you see a case where solverDataDict can contain different soft types.

GearsAD, 2019-11-24:
There isn't really a strong reason to populate it, to be honest. I believe it's populated because before softtype specialization and the solver dictionary, there was always a value. I _think_ if we remove the default value then things will blow up (a lot of code just expects a default value to be there), but we can clean that up if it's important to your testing?

Affie, 2020-02-13:
I removed it and nothing broke in IIF tests. `addVariable!` overwrites :default (and :parametric) if its there.

commit 587bc225 (2020-02-13): Merge pull request #298 

Don't create :default => empty VND with constructor close #202
closed 2020-02-13 by Affie