JuliaRobotics/DistributedFactorGraphs.jl #211: Variables and factors performance and parametric types

Affie, 2019-11-24:
Relating to performance tips:

1. [Avoid containers with abstract type parameters](https://docs.julialang.org/en/v1/manual/performance-tips/index.html#Avoid-containers-with-abstract-type-parameters-1)
With the current data design we don't have a choice. Not all entries will have the same type (softtype). But it doesn't seem to matter much in this case. The really bad for performance case is when you have for example: a `Vector{Real}` and you mix `Int`, `Float64` etc it is forced to store a reference to the value and not the value directly. 
2. [Avoid fields with abstract type](https://docs.julialang.org/en/v1/manual/performance-tips/index.html#Avoid-fields-with-abstract-type-1) Not exactly an abstract field in this case, but it still creates a  type instability seen with `@code_warntype`.
Effect can be minimized with [function barriers](https://docs.julialang.org/en/v1/manual/performance-tips/index.html#kernel-functions-1)

Related see PR #199, PR #200. Issue #149, #147
comment on [@code_warntype](https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/148#issuecomment-546993005)

#### `DFGFactor`
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/a1f16da2c433a851ef516efe7a11c4b3ce64be02/src/entities/DFGFactor.jl#L37-L47

```julia
@code_warntype getFactor(fg, :x0l1f1)
```
[image]

#### `DFGVariable` https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/a1f16da2c433a851ef516efe7a11c4b3ce64be02/src/entities/DFGVariable.jl#L111-L122

Two places that introduce potential type instabilities: `solverDataDict` and the variables dict in dfg itself `variables::Dict{Symbol,V}`,
- Variables dict: like previously said, we can't get rid of it completely as not all variables are the same type (softtype). I don't believe this is a serious problem because of  function barriers and the way it is used.

- `solverDataDict`  can be typed with the design in PR #199 to be similar to `DFGFactor`, It might have rather big downstream impact though, (I haven't checked). 

https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/238116f289d03b91f2fdf63c5e6425cf7e7be480/src/entities/DFGVariable.jl#L109-L114

So the bottom line: 

- [ ] Keep as it is now, or

- [ ] Parametric type  `solverDataDict`  as in PR #199

GearsAD, 2019-11-24:
Awesome - thanks. I'm leaning towards #199 and checking what happens downstream. If I understand correctly, the only limitation is that once we cast the `solverDataDict`, we can't have different types in it. I'm a little worried we may want different types for different solves (e.g. supersolve). @dehann is that a viable use-case?

dehann, 2019-11-25:
Ah @Affie , this is awesome thanks!!

dehann, 2019-11-25:
HI @GearsAD , regarding different softtypes for different supersolves -- i think that is a no-go.  There is a point where you need to build a whole new graph (duplicate session or whatever).  My current intuition is that supersolve is for numerical variations.

That said, if we want to collapse long portions of the factor graph as a variable reduction thing we need to be careful about where we draw the line.  In this reduction case we would still be using the same softtype for each supersolve, but will be using different factors for each of the cases.  That is okay I think... Design will just be interesting.

dehann, 2019-11-25:
add for later reference:
https://julialang.org/blog/2018/08/union-splitting

Affie, 2019-11-27:
See also warn type for GraphsDFG: 
[image]

GearsAD, 2019-11-27:
Will look on the weekend :+1: I think it may be the same case for all drivers, not just GraphsDFG?

Affie, 2019-11-27:
Yes first one was with LightDFG, all will have the DFGFactor instability. Just as information, I don't think it is a problem as long as it is not propagated to far.

Affie, 2020-02-07:
@GearsAD  implemented Parametric type solverDataDict as in PR #199 in his 0.6 branch.  
I would just like to confirm we can close this as resolved with:
- Parametric type solverDataDict as in PR #199

dehann, 2020-02-10:
I think this can be closed yes -- not sure I'm understanding it right, but the static parameter for variable {T} is not too bad.  If that is what you asking.

Affie, 2020-02-12:
Implemented in #291

closed 2020-02-12 by Affie