JuliaRobotics/DistributedFactorGraphs.jl #214: Mini cleanup on solvable for GetSubgraphAroundNode on LightGraphs

GearsAD, 2019-11-25:
Hi @Affie,

I added this https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/master/src/LightDFG/services/LightDFG.jl#L395 - but I think it will behave differently than the other drivers because it doesn't grow from 1 to `distance` neighbors. E.g. if in a graph x0 is solvable, x0x1f1 is not solvable, and x1 is solvable, asking for a subgraph around x0 with only solvable will return x1. 

I'm checking and adding a test for it, but do you mind taking a look at whether we can change it if needed?

GearsAD, 2019-11-25:
Branch with test that shows it doesn't maintain continuity - https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/compare/bug/214_subgrapharoundnode

Affie, 2019-11-25:
Hi, looks like we have different approaches on getting the subgraph.
It will depend if it should be a connected sub graph or not. I get the entire sub-graph and then filter it. So it can have multiple fragments.

Related, why should the node you start with always be returned even if its not solvable?

GearsAD, 2019-11-25:
I'm worried about fragmented graphs, this has been a problem in the past. There's probably no hard and fast rule, but the idea of growing+filtering means that you will get a connected subgraph with all solvable.

For the first node, either can or can't return, but it's a boundary case - if you ask for a node that will be filtered then you're guaranteed to have a fragmented graph (strange user request).

GearsAD, 2019-11-25:
@dehann - do you have a strong preference either way? This is pretty much the only thing holding up 0.5.1.

Let there be a graph x0 -> x0x1f1 -> x1 -> x1x2f1 -> x2, and everything is SOLVABLE except x1. 
If I ask for getSubgraphAroundNode(..., x0x1f1, distance=2, solvable=1), then I could one of two results:
1. Solvable is not cascaded so I get a fragmented graph [x0 -> x0x1f1, x1x2f1 -> x2]
2. Solvable is cascaded, so I get [x0 -> x0x1f1]

No strong preference my side, I like the idea of connected graphs so I like 2, but that may just be because of past issues with fragmented graphs.

dehann, 2019-11-25:
yes my preference is a default with keyword argument to either selfCheckSolvable or nog.

fragmented graphs are hard to work with sometimes.

i would also think it is on the user to know what they are doing.  there will always be a better fool out there so fool proofing only the obvious stuff makes sense.

there i think default should be the way Sam has it, but if the other case were a problem then the user should be able to ask for self check.

dehann, 2019-11-25:
also, user asking for a self check is the same as the user checking themselves.

dehann, 2019-11-25:
i’m with Sam on this one.

Affie, 2019-11-25:
Sam wrote both, so I guess you mean 2? I don't have an opinion.  GraphsDFG and SymbolDFG is currently 2 and LightDFG and CloudDFG is 1.

Affie, 2019-11-25:
What about the other way around, If you want to get all nodes close to :x1 that is not solvable to change them into solvable? Or is that where `ls` comes in.

dehann, 2019-11-25:
my suggestion is to ignore the solvable flag from the requesting node as default.  if it’s expected to be a bigger issue then add a warning.

i’m worried that dfg has too much maintenance load.  we need to reduce the aperture there by spending time on better using dispatch or reducing the amount of code.  taking out Metagraphs already helps.  SymbolDFG is a bit excessive at the moment but let’s try fold it in as much as possible.

Cloud, Graff, File, light, graphs also need to be maintained — so the maintenance problem is only going to get worse.

general request:  if anyone sees
something that can be concentrated through multiple dispatch on abstracts, please open an issue and ask for help.

dehann, 2019-11-25:
solvable is user dictated so we cannot change it for filtering purposes.

GearsAD, 2019-11-27:
@Affie could you change lightgraphs to match the pattern please?

GearsAD, 2019-11-27:
I'll do the same for CGDFG then.

Affie, 2019-11-27:
It can maybe be copy and paste of SymbolDFG, does SymbolDFG work as you would like?

Affie, 2019-11-28:
I had a look at it and have a bit more of a general philosophy question. We have 2 choices:

1. write and maintain it our self as in GraphsDFG, SymbolDFG
    - The more we rewrite basic graph functionality the more we diverge form LightGraphs.jl. The motivating reason to stick to LightDFG and not SymbolDFG was to use existing functions. LightDFG and CloudDFG works the same, and are both externally maintained. 

2. use existing functions,  such as  `connected_components(g)` in LightGraphs.jl  
    - so I can maybe add a parameter in _copyIntoGraph that only copies connected components

Affie, 2019-12-04:
Basics of how 2 can work for future self.
```julia
ns = neighborhood(fg.g, fg.g.labels[:x3], 3)
#solvable != 0 && (filter!(id -> _isSolvable(dfg, dfg.g.labels[id], solvable), ns))
#if copy only connected subgraph - look for it in connected components
h,vmap =  induced_subgraph(fg.g.graph, ns)
cc = LightGraphs.connected_components(h)
```

Currently `GetSubgraphAroundNode` is only used in IIF at one place `buildSubgraphFromLabels`, where the distance is 2 just to make sure all factors are included. The extra variables and orphaned factors are then removed. The cliqSubFG is mostly a disconnected graph.

dehann, 2020-01-21:
>  The cliqSubFG is mostly a disconnected graph.

After the big review of Bayes tree stuff, it makes sense to keep cliqSubFG as fragments, definitely not advisable to fill in connecting fragments between.

> Currently GetSubgraphAroundNode is only used in IIF at one place buildSubgraphFromLabels

Jip, currently -- but we don't want DFG to be a silo'ed set of wrappers around outside libraries.  We want the minimal set of code to easy cross feed between the various DFG systems.  This is a deeply fundamental objective of DFG.  I made a slight modification to the repo slogan to try keep focus on that:
```
Abstraction layer for cross-feeding distributed factor graphs over various technologies 
```

Also want to reiterate an eariler comment: 
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/214#issuecomment-558186644

dehann, 2020-01-21:
> so I can maybe add a parameter in _copyIntoGraph that only copies connected components

Lets get a common version that works for everybody in AbstractDFG stabilized first

> does SymbolDFG work as you would like?

Are we still going to use SymbolDFG, I unfortunately can't remember?  Sounds like GraphsDFG and LigthDFG are the major in memory flavors for the time being?

dehann, 2020-01-21:
How far away are we from closing this issue?  Is #261 and the ensuing "cross-feeding" vs "always performance" --I'd like to separate out the design philosophy here please: #264

PR #263 (2020-01-22): Abstract getSubgraphAroundNode
Close #214
Close #261
closed 2020-01-22 by GearsAD