JuliaRobotics/DistributedFactorGraphs.jl #222: How do you filter to only return solvable == 0

Affie, 2019-11-27:


GearsAD, 2019-11-27:
As per #201:
> Using solvable in this way means that it's a lower bound when filtering - filtering on solvable=0 means that you'll get both solvable=1 and solvable=0. That is correct?
> DF correct

At present there isn't a way to query unsolved nodes.

Affie, 2019-11-28:
Ok, I missed the reasoning behind it. What about a compromise. 
  - use an `Array{Int} = []` or 
  - use an `UnitRange{Int64}`

eg.
```julia
function _isSolvable(dfg::LightDFG, label::Symbol, ready::Array{Int})::Bool
	haskey(dfg.g.variables, label) && (return dfg.g.variables[label].solvable in ready)
	haskey(dfg.g.factors, label) && (return dfg.g.factors[label].solvable in ready)

...
```

GearsAD, 2020-02-08:
Thinking about this - if filtering solvable==0 is needed, I'd suggest we don't use the lower limit then. Rather we switch it to an equality check than introduce the array?

Affie, 2020-02-24:
I came across another case here #331. Maybe `==` will work better as you said. However, I don't understand the reason behind `>` so hesitant to change.

dehann, 2020-04-27:
Think we resolved this in a previous discussion?  The main idea is/was that there are levels of solvable.  The way I find `solvable == 0` is to use `setdiff(ls(fg,solvable=0), ls(fg, solvable=1))`.  Its working well enough for the time being that we can close this issue?

dehann, 2020-05-02:
> understand the reason behind > so hesitant to change.

We need a way to search for multiple values of solvable at once, and rather than doing a ton of `x in list` it seemed more sensible to work with an Int and `<` or `>`.  So `ls(fg, solvable=0)` will include all `0 <= val` -- i.e. ==0, ==1, ==2, etc.

The easiest way then is to use `setdiff` for only getting `==0` but excluding `==1` or `==0` cases.  However, this way is easier to get just solvables and skip unsolvables.

Affie, 2024-10-21:
I propose we allow users to choose with a `by::Function = >` kwarg, supporting `[<,>,<=,>=,==]`

PR #1097 (2024-10-22): solvableFilter on listVariables and getVariables for GraphsDFG
TODO: still need to be added to all relevant functions and tests.

close #222
close #331