JuliaRobotics/DistributedFactorGraphs.jl #229: [FEATURE] addTags! function for post-hoc addition

dehann, 2019-12-02:
This might also involve IIF somehow, but it would be great to be able to add tags through the API after a variable already exists, e.g.:
```julia
addTags!(fg, :x4, [:RADAR; :LIDAR])
```

dehann, 2019-12-02:
On second thought, that might mean full CRUD on tags?

GearsAD, 2019-12-04:
Yes, but updateVariable should update the tags, I would suggest going that way for now. Would that work?

dehann, 2019-12-19:
This issue also requires the common definitions for api or CRUD to be known.  Please add comments on the `verbNoun` definitions wiki here (link via docs):
https://www.juliarobotics.org/Caesar.jl/latest/dev/wiki/#Standardizing-the-API,-verbNoun-Definitions:-1

PR #291 (2020-02-12): Development updates for DFG v0.6
partial fix for #241 
fix #239
close #229
implement #211
fix #172
part of #234
closed 2020-02-12 by Affie