JuliaRobotics/DistributedFactorGraphs.jl #232: [API] request name change of function timestamp -> getTimestamp

dehann, 2019-12-06:
Hi,  could we perhaps use the name please:
```julia
getTimestamp(vx) = vx.timestamp
```
I had a situation here where Julia gave a valid error but not at all related to what was going on.  Took a while, and then found the mistake was I was using a variable called timestamp (which would likely occur often) which Julia was trying to call as a Functor.

dehann, 2019-12-06:
xref #230

dehann, 2019-12-07:
I think the names are nice but unfortunately the overriding names create a few problems with unexpected error messages.  Boring accessor and set functions are more predictable?

Affie, 2019-12-07:
I agree that is is likely for someone to have a variable tags or timestamp. Always using `DFG.timestamp()` can maybe solve this, but it seems a bit tedious. 
Do you propose we should change all existing accessor functions? A few other with potentially the same issue.  
- label
- estimate - WARNING: both DistributedFactorGraphs and Distributions export "estimate"; uses of it in module IncrementalInference must be qualified
- softtype

It should be noted that most of them returns a reference to the objects and ideally one would not want to use set as it would create new memory.

dehann, 2019-12-07:
[image]

dehann, 2019-12-07:
yes, I am proposing that all accessor function are always `getX`,  this would be consistent with the rest of the IIF/KDE/AMP/RoME/Plotting api.

Similarly, also always `setX!`.

Guess we should compile a more standard language here -- something that is consistent and user guesses should be right as frequently as possible.  Something like:
- functions are `verbNounSpecial`
  -  `getVariable`, `getVariableIds`, `getLabel(<:AbstractVariable)`
  - `getMarginalBelief(<:AbstractDFG, ::Symbol)`
  - `isInitialized!`, `findVariableNearTimestamp`, `findFactorNearTimestamp`

Thinking out load, we can then define standard verbs and nouns (thats nothing new) but then populate the matrix between them as is sensible.

**verbs**
- `get`
- `set`
- `find`
- `is`
- `list` / `ls` / `lsf`
- `build`

**Nouns**
- `Variable`
- `Factor`
- `Timestamp` / `Tags`

**Please** edit and add any we can think of here -- which we can add to general documentation.

Affie, 2019-12-08:
I’m onboard for get- methods and standardio.  See also xref #233.

dehann, 2019-12-16:
Alright, moving the definition discussion to wiki here:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/wiki/Standard-Definitions-of-API-via-verbs-and-set-theory

dehann, 2019-12-16:
definitions are the best way out I see at the moment -- proper ratification process is required and any concerns must be listed on the wiki.  Going to take a few weeks to do that properly, then hopefully API is easier

commit ba710149 (2019-12-06): fix #232
closed 2019-12-17 by GearsAD