JuliaRobotics/DistributedFactorGraphs.jl #233: Feature request: Add field methods in type docstrings.

Affie, 2019-12-07:
How would you feel about adding the filed accessor methods in the type doc strings?
I constantly find myself having to go search for the methods and end up rather using the `.` notation.

eg
```julia
help?> SkeletonDFGFactor
  struct SkeletonDFGFactor <: AbstractDFGFactor

  Skeleton factor with essentials.

    •    label
      
        Factor label, :x1f1, label(factor)

    •    tags
      
        Factor tags, [:FACTOR], tags(factor)

    •    _variableOrderSymbols
      
        The order of the variables associated with this factor.
```

Then I would know to rather use `label(factor)` than `factor.label`

Affie, 2019-12-07:
What causes additional complications in this case is that 
```julia
getVariableOrder(fct::DFGFactor)::Vector{Symbol} = fct._variableOrderSymbols
```
is defined in IIF.

dehann, 2019-12-10:
Im suggesting in #232  that we use `getTags/getLabel` rather than `label/tags` as you wrote above.  I have not used the field methods in docstring before, are you talking about this?
https://juliadocs.github.io/DocStringExtensions.jl/latest/#DocStringExtensions.FIELDS

Affie, 2019-12-10:
Yes. Its just an example. https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/068a2fed964440edcbd17e6346bfd65469cacbeb

dehann, 2019-12-16:
Yes, I like the docs this way -- good idea thanks!

dehann, 2019-12-16:
pending standardization of the actual function calls, `getTimestamp/getLabel` etc

GearsAD, 2019-12-18:
Sure, that looks good!

To be fair, I'm not really against the dot notation personally, accessors or dot notation they both have issues - the only time the accessors are markedly better is when you have to hunt something down and there's logic in the actual accessor (e.g. `getSolverData(variable, :default)`.

Affie, 2020-02-07:
Seems everybody is happy with doing it. It is already implemented in a few types. 
We should maybe write it down somewhere in a developer guide.

- [ ] start developer guide

closed 2020-02-12 by Affie