JuliaRobotics/DistributedFactorGraphs.jl #234: [FEATURE] filter on tags

dehann, 2019-12-07:
Hi, not sure what the plan is around convenience functions, but perhaps we should look at supporting tag based searches as a general thing.  Not sure if this should be a secondary user call, or integrated with the general functions like `ls/lsf` from the get go?

I just came across a case where this would be mighty convenient -- check if any of the attached factors have the MAGNETOMETER tag:
```julia
ls(fg, :x1, tags=[:MAGNETOMETER;])
```

The idea in this case is to quickly filter out whether a particular type of tag has already been inserted (Prior factor in this case) before, which simplifies front end logic significantly.

dehann, 2019-12-07:
also see #232

dehann, 2019-12-07:
think we doubled up `ls` and `lsf` in this case---believe both should work---not sure if we need to be more specific here -- please comment if there are failure cases.

dehann, 2019-12-07:
also both Variables and Factors can have tags.

Affie, 2019-12-10:
That should already be possible:
```julia
function ls(dfg::G, regexFilter::Union{Nothing, Regex}=nothing; tags::Vector{Symbol}=Symbol[], solvable::Int=0)::Vector{Symbol} where G <: AbstractDFG
    return getVariableIds(dfg, regexFilter, tags=tags, solvable=solvable)
end
```
```julia
function getVariables(dfg::LightDFG, regexFilter::Union{Nothing, Regex}=nothing; tags::Vector{Symbol}=Symbol[], solvable::Int=0)::Vector{AbstractDFGVariable}

    # variables = map(v -> v.dfgNode, filter(n -> n.dfgNode isa DFGVariable, vertices(dfg.g)))
    variables = collect(values(dfg.g.variables))
    if regexFilter != nothing
        variables = filter(v -> occursin(regexFilter, String(v.label)), variables)
    end
    if solvable != 0
        variables = filter(v -> _isSolvable(dfg, v.label, solvable), variables)
    end
    if length(tags) > 0
        mask = map(v -> length(intersect(v.tags, tags)) > 0, variables )
        return variables[mask]
    end
    return variables
end
```

Affie, 2019-12-10:
Sorry I see you actually mean filter the neighboring factors of a variable?
So is it actually `function getNeighbors(dfg::LightDFG, node::DFGNode; solvable::Int=0)::Vector{Symbol}` where you want the additional filters?

Affie, 2020-02-13:
see #291 for tags filter added to `getFactors`, it just has to be extended to the rest of the functions

dehann, 2020-02-14:
HI @Affie , sorry yes -- `getNeighbors`, perhaps also `listNeighbors` with `tags` filter is what I'm looking for.  Will add issue to Caesar.jl to better document this there, thanks!

closed 2020-02-14 by dehann