JuliaRobotics/DistributedFactorGraphs.jl #237: [DEPRECATE] `softypename::Symbol` was a bad idea

dehann, 2019-12-16:
Just keep it as a `softtype` in both `DFGVariable` and `DFGVariableSummary`?

dehann, 2019-12-16:
FYI @GearsAD , I responded here and updated regarding your earlier comment:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/wiki/Standard-Definitions-of-API-via-verbs-and-set-theory

Affie, 2019-12-24:
Can you elaborate why it was a bad idea?

#199 will give a type to DFGVariable but how will we handle DFGVariableSummary. 

How does softtype fit in with possible future use of types such as quaternion on manifold.

dehann, 2020-02-02:
Depends on what the user wants to do, if the purpose of DFGVariableSummary is just tell you, hey this is Pose2 but doesn't tell you what it is -- okay.  If you need all the manifold information then ... you will need to have a definition of Pose2 to deserialize.

@Affie , perhaps we can scope what Summary is supposed to do?

Affie, 2020-02-02:
One use case is visualization and plotting and PPE is worthless without a type.

Affie, 2020-02-07:
What is the replacement? I think we should maybe leave for later

Affie, 2020-02-11:
Somewhat related, is factor type needed in summary?

Affie, 2020-02-14:
I think I see where the problem is:
`getSofttype(v::DFGVariableSummary)::Symbol = v.softtypename`
accessor should be `getSofttypename`. 
`getSofttype` can try build the original softtype, but we should first build a use case around it.

Affie, 2020-02-15:
See https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/6b903b79b15c6754191cfb2b673dd89d1eb97e2c/src/services/DFGVariable.jl#L290
And https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/6b903b79b15c6754191cfb2b673dd89d1eb97e2c/src/services/DFGVariable.jl#L293

Affie, 2020-04-15:
@dehann, can I close this? or do you still have a problem with `softtypename`?

dehann, 2020-07-08:
i think we should solve @GearsAD ‘s concerns regarding softtype on CG first before we close this?  I’m happy with name only if it solves the right problem.  I’m worried about the CG stuff is going down a Gaussian-only road and end up forcing the overall DFG side to a crisp but overly restrictive DataLevel1 space.  I mean that users just want Level1 and moan it’s not working well enough, and somehow get out of using Level2.  To help avoid all that i think softtypename::Symbol/String is good.

Just to reiterate, i originally wanted to avoid duplication of the whole softtype so just guessed name as a string as cheap hack.  Without the necessary type information for non-Julia use cases, i’m not sure... Again the question for me is what do we want from Summary (what is the one sentence existence proof for Summary/Level1?).

Is it fair for a user to avoid the Julia internals just to get the inference product in a usable way, or should they at least don some of the Julia-ness to use the inference products?  Perhaps i should add, (again) that we are not building Gaussian-only, so the user should get the full beliefs, not just hide behind PPE.  That means DataLevel2.  I don’t want to dumb down the output and look shoddy against a more mature Gaussian-only system.  I’d rather force the user into Julia or build a better multilang interface (I vote the latter). 

I know Neo4j has a bunch of nice features for Gaussian-only, but that should be “supportive” not “authoritative” in the design.

I would argue Summary/Level1 CANNOT be it is the final inference product.  The final inference product is and always should be DataLevel2.

TL;DR I fairly strongly feel softtypename as Symbol/String or remove completely from Level1.  If there are cool features in Neo4j we can leverage, then leave that door open;  but the DB must never dictate our design for what we want to store, DB must store what is needed for the problem — problem here is *strongly* non-Gaussian (ie not just PPE).

Affie, 2020-07-08:
I understand your concern, but I don't think softtype will make much of a difference. At the end (with a robot) users are only interested in its pose or possible poses and perhaps the covariance.  I think improving PPE's will help with this. All the modes and covariance will help, but then perhaps we can add a "Gaussian mode fit metric". If its close to 1, you can safely use only the PPE, but if its close to 0, it is far from Gaussian and PPE cannot be used.

Affie, 2020-08-10:
With the change of softtypes to singleton types, softtypename becomes more useful. Should just add the module name see #157.

closed 2020-08-10 by Affie

dehann, 2020-08-11:
Great thanks!