JuliaRobotics/DistributedFactorGraphs.jl #239: [BUG] TYPEDFIELDS did not work right

dehann, 2019-12-18:
I had to do quick removal on master here: 34b7468  on account of fresh install error:

```julia
julia> using IncrementalInference
[ Info: Precompiling IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480]
[ Info: Multithreaded  convolutions possible, Threads.nthreads()=1.  See `addFactor!(.;threadmodel=MultiThreaded)`.
WARNING: Method definition (::Type{DistributedFactorGraphs.VariableNodeData{T} where T<:DistributedFactorGraphs.InferenceVariable})(Array{Float64, 2}, Array{Float64, 2}, Array{Symbol, 1}, Array{Int64, 1}, Int64, Bool, Symbol, Array{Symbol, 1}, T<:DistributedFactorGraphs.InferenceVariable, Bool, Float64, Bool, Bool, Int64) where {T<:DistributedFactorGraphs.InferenceVariable} in module DistributedFactorGraphs at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/entities/DFGVariable.jl:11 overwritten at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/entities/DFGVariable.jl:41.
  ** incremental compilation may be fatally broken for this module **ERROR: LoadError: LoadError: UndefVarError: TYPEDFIELDS not defined
Stacktrace:
 [1] top-level scope at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/entities/DFGVariable.jl:123
 [2] include at ./boot.jl:328 [inlined]
 [3] include_relative(::Module, ::String) at ./loading.jl:1094
 [4] include at ./Base.jl:31 [inlined]
 [5] include(::String) at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/DistributedFactorGraphs.jl:1
 [6] top-level scope at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/DistributedFactorGraphs.jl:16
 [7] include at ./boot.jl:328 [inlined]
 [8] include_relative(::Module, ::String) at ./loading.jl:1094
 [9] include(::Module, ::String) at ./Base.jl:31
 [10] top-level scope at none:2
 [11] eval at ./boot.jl:330 [inlined]
 [12] eval(::Expr) at ./client.jl:432
 [13] top-level scope at ./none:3
in expression starting at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/entities/DFGVariable.jl:123
in expression starting at /home/tonio/.julia/packages/DistributedFactorGraphs/NMxiR/src/DistributedFactorGraphs.jl:16
ERROR: LoadError: Failed to precompile DistributedFactorGraphs [b5cc3c7e-6572-11e9-2517-99fb8daf2f04] to /home/tonio/.julia/compiled/v1.2/DistributedFactorGraphs/BWCpj.ji.
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] compilecache(::Base.PkgId, ::String) at ./loading.jl:1253
 [3] _require(::Base.PkgId) at ./loading.jl:1013
 [4] require(::Base.PkgId) at ./loading.jl:911
 [5] require(::Module, ::Symbol) at ./loading.jl:906
 [6] include at ./boot.jl:328 [inlined]
 [7] include_relative(::Module, ::String) at ./loading.jl:1094
 [8] include(::Module, ::String) at ./Base.jl:31
 [9] top-level scope at none:2
 [10] eval at ./boot.jl:330 [inlined]
 [11] eval(::Expr) at ./client.jl:432
 [12] top-level scope at ./none:3
in expression starting at /home/tonio/.julia/dev/IncrementalInference/src/IncrementalInference.jl:15
ERROR: Failed to precompile IncrementalInference [904591bb-b899-562f-9e6f-b8df64c7d480] to /home/tonio/.julia/compiled/v1.2/IncrementalInference/NrVw2.ji.
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] compilecache(::Base.PkgId, ::String) at ./loading.jl:1253
 [3] _require(::Base.PkgId) at ./loading.jl:1013
 [4] require(::Base.PkgId) at ./loading.jl:911
 [5] require(::Module, ::Symbol) at ./loading.jl:906
```

found by @tonioteran -- thanks!

tonioteran, 2019-12-18:
my pleasure! ;)

GearsAD, 2019-12-18:
Nice! Thanks

GearsAD, 2019-12-18:
TYPEDFIELDS in DocstringExtensions >= 0.8. Bumped up the minimum version on master compat for this.

Affie, 2020-02-07:
Looks like bump somehow got reverted, will be implemented in v0.6

GearsAD, 2020-02-08:
This is resolved in the branches and I think @dehann you checked all is ok downstream - ok to close?

PR #291 (2020-02-12): Development updates for DFG v0.6
partial fix for #241 
fix #239
close #229
implement #211
fix #172
part of #234
closed 2020-02-12 by Affie