JuliaRobotics/DistributedFactorGraphs.jl #252: Precompile warning on DFGVariable

dehann, 2019-12-30:
```julia
WARNING: Method definition (::Type{DistributedFactorGraphs.VariableNodeData{T} where T<:DistributedFactorGraphs.InferenceVariable})
(Array{Float64, 2}, Array{Float64, 2}, Array{Symbol, 1}, Array{Int64, 1}, Int64, Bool, Symbol, 
Array{Symbol, 1}, T<:DistributedFactorGraphs.InferenceVariable, Bool, Float64, Bool, Bool, 
Int64) where {T<:DistributedFactorGraphs.InferenceVariable} 
in module DistributedFactorGraphs at /home/travis/.julia/packages/
DistributedFactorGraphs/G2Wc2/src/entities/DFGVariable.jl:11 overwritten at /home/travis/
.julia/packages/DistributedFactorGraphs/G2Wc2/src/entities/DFGVariable.jl:41.

  ** incremental compilation may be fatally broken for this module **
```

GearsAD, 2020-01-02:
Ah, yeah - that is the constructor. Don't think it's serious, but will try include a fix for 0.6

Affie, 2020-02-12:
I haven't seen this, so it might be fixed by #291

closed 2020-02-13 by Affie