JuliaRobotics/DistributedFactorGraphs.jl #264: [DESIGN] "cross-feeding" vs "performance" -- ie dispatch vs specialized

dehann, 2020-01-21:
DFG is about having access to various graphing technologies in a way that allows info/data/etc to be easily transferred between technologies in a standard way.  For example:

```julia
merge!(CloudDFG, FileDFG)
merge!(FileDFG, CloudDFG)
merge!(CloudDFG, LightDFG)
merge!(GraphsDFG, LightDFG)
merge!(GraffDFG, CloudDFG) # seems excessive, but its about the philosophy 
# ...

update!(LigthDFG, CloudDFG)
update!(FileDFG, GraphDFG)
# ...

getSubGraph(::LightDFG, GraphDFG)
getSubGraph(::CloudDFG, LightDFG)
getSubGraph(::FileDFG, CloudDFG)
# ...
```

Originates from hard discussions on:
- #261 
- #214

dehann, 2020-01-21:
Decision is whether we want to support the kinds of "cross-feeding" functions I'm suggesting here -- or which subset?  Or, if any decisions are going to actively exclude any of the features suggested here?

dehann, 2020-01-22:
In reponse to a discussion in #261:

This is more speaking to the room:  I want to stress that DFG should not just be a clambering together of silo'ed drivers for graph technologies -- the most value comes from true "Distributed" nature in a variety of workloads that one user probably wont' see in their individual use-cases.  I fully acknowledge its hard to get right though.

dehann, 2020-05-24:
would you guys say the current design in DFG v0.8.0 achieves this -- so we can close?

Affie, 2020-05-26:
It’s definitely a step in that direction. I think it’s going to be a continuous effort.

dehann, 2020-06-14:
Cannot close until at least #507 is resolved

dehann, 2020-07-26:
https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/issues/315#issuecomment-661027817