JuliaRobotics/DistributedFactorGraphs.jl #266: [REFACTOR] move InMemDFGType here from IIF

dehann, 2020-01-22:
https://github.com/JuliaRobotics/IncrementalInference.jl/blob/master/src/IncrementalInference.jl#L526-L527

Affie, 2020-02-07:
It is the type used by IIF and does not belong here. 
I would not want to modify DFG to work with another type when doing development in IIF.

GearsAD, 2020-02-08:
I view it as just a definition of what DFG currently offers as in-memory drivers, why do you feel that IIF should keep that list? I'd think IIF would just consume it.

Affie, 2020-02-10:
> I view it as just a definition of what DFG currently offers as in-memory drivers, why do you feel that IIF should keep that list? I'd think IIF would just consume it.

That's `InMemmoryDFGTypes` (already in DFG and a good fit)

> It is the type used by IIF and does not belong here.
> I would not want to modify DFG to work with another type when doing development in IIF.

`InMemDFGType` : Default used by dfg, to be fixed to LightDFG once Dehann is happy with  `LightDFG` as default.

Sorry for my poor choice with names.

Can I close?

dehann, 2020-02-12:
@Affie just make sure you have the spelling right `InMemoryDFGTypes`

Affie, 2020-02-13:
If it is set to `LightDFG` then I guess it's ok, don't understand why you want to move it though? It is not used at all in DFG and you can break IIF if you accidentally change it.

GearsAD, 2020-02-16:
Ah, i see - this isn't the list, it's the actual chosen type. In that case I think it probably should live in IIF.

dehann, 2020-02-17:
2 votes to stay in IIF and 0 votes to move to DFG then.

closed 2020-02-17 by dehann