JuliaRobotics/DistributedFactorGraphs.jl #273: [FEAT] add Condition field to VND

dehann, 2020-01-28:
Lets user block on a variable but immediately use the result after a notification (likely downsolve update), regardless of how far the solver has progressed.

dehann, 2020-02-12:
updated comment above

Affie, 2020-02-13:
I think its a bit more complicated than just adding the field. I added commit to see how it would work https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/61a988622964dc4501a03b74478093f2e7d84f8e

I'm wondering if a `Dict{Symbol, Condition}` (perhaps in the tree) will not be a better place to sort out the functionality first. If we add the field functions such as: compare, pack, copy get complicated and has to be reworked. 
Take the use case for example, IIF currently makes subgraph deepcopies. So the conditionals will not work. The subgraph can also exist on a different core/computer.

dehann, 2020-02-19:
Can we skip the condition field during packing, and just create a new one when unpacking?

Also, do you know if `Condition` and `Thread.Condition` are interchangeable?

dehann, 2020-02-19:
We should probably only use Threads.Condition and therefore lift our minimum Julia requirement to 1.2:
https://docs.julialang.org/en/v1/base/parallel/#Base.Threads.Condition

PR #310 (2020-03-03): Add event::Threads.Condition to VND
fixes #273
closed 2020-03-03 by dehann