JuliaRobotics/DistributedFactorGraphs.jl #274: GenericFunctionNodeData{T, S} T and S types

Affie, 2020-01-31:
What type is T and can we inherit so its not any?
`fnc::T`

What is `frommodule`?, can it not be removed with new serialization in DFG or at least fixed to a `Symbol`
frommodule::S #Union{Symbol, AbstractString}

dehann, 2020-02-02:
Hi @Affie, this is similar to `softtype` for variables, but here its a hard type.  Can be `Pose2` or `InertialPose3`, `Point2`, `ContinuousScalar` etc.

dehann, 2020-02-02:
I'm not following on the `frommodule` part?

dehann, 2020-02-02:
Id say don't fuss with this until DFG v0.7.  Too many things are busy changing and might be quite hard to migrate all the downstream code so quickly.

Affie, 2020-02-02:
I would just like to fill in the type (so its not `Any`) and understand the code. 
I did a search in dfg and iif for `frommodule` and couldn’t figure out what it is used for.

GearsAD, 2020-02-03:
Looking at the code, at least from usages we can just initialize a GenericFunctionNodeData{T,S} to be GenericFunctionNodeData{T}, and let frommodule be a symbol. I don't think it's used anywhere else in any form.

GearsAD, 2020-02-03:
Ah, there's a couple places in IIF where GenericFunctionNodeData is used. This would need some downstream cleanup.

dehann, 2020-02-03:
I think T will be something like `FunctorInferenceType` or similar, cannot entirely remember right now.  If you confident about it and tests pass then do it on a PR and we target that for DFG v0.6 then too.  If you eager for the clean-up then we should probably just do it.

I'm not attached to frommodule, so if its not serving any purpose thats okay to go.  Feel free to decide whats best there.

Affie, 2020-02-06:
Hi, I'm still not confident, however it is something we need to do. It simplifies a few things.
From what I can see it is a `<:ConvolutionObject{<:FunctorInferenceType}`, but would like to get confirmation.

dehann, 2020-02-12:
I can confirm this right, use ConvolutionObject as abstraction to fortify types and fields

Affie, 2020-02-13:
Thanks Dehann, it helped for my tests. 
It looks like its used in the packed types also as `PackedInferenceType`. Will one day follow the breadcrumbs.

dehann, 2020-02-20:
> Since Symbol <-> String conversion exists, It can really just be of type Symbol.

I think we needed string for storage to file at some point in time.  Not sure if Symbol will autoconvert on current code

dehann, 2020-04-24:
@Affie , I think we resolved this on the phone call today:
- [ ] lets plan to deprecate `frommodule`... after the `ConvolutionObject` abstract consolidation 
  - [ ] think you already halfway busy with `ConvolutionObject` --> `FactorOperationalMemory` ?

Affie, 2020-04-24:
Yes. Thanks for helping me understand it. I’ll try and find an nice way to do the `ConvolutionObject` --> `FactorOperationalMemory` rename change, but it will probably be a breaking change. So lets do that in v0.8. 
For now I’ll fill in the types and get the interface  tests running without iif.

closed 2020-05-06 by Affie