JuliaRobotics/DistributedFactorGraphs.jl #281: Check Cloud updateFactor!

Affie, 2020-02-07:
Confirm cloud updateFactor! works correctly with reduced interface.

Affie, 2020-02-07:
What should happen if you update a factor with by adding new variables to it?
Surely its not the same factor any more and the label shouldn't be the same.
The same goes for changing the variables assigned to it.

Affie, 2020-02-07:
Should `updateFactor!` error if `variableOrder` is changed with a message to delete it and add a new one.
So fields to be seen as "immutable":
- `label` 
- `variableOrder` 
- `timestamp`
- Its type

GearsAD, 2020-02-10:
I took a look and in general CGDFG needs a significant refactor, there's a draft PR #285 of the template for the functions.

Affie, 2020-02-14:
> yep, we just need to add validation to all the updates to error if variables are different
<sub>[View in Slack](https://caesarjl.slack.com/archives/CKGRFPZT8/p1581695107023000)</sub>

GearsAD, 2020-02-17:
Removed that function and put the check in - https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/pull/285/files#diff-60feee61df5ad89116b72980e992910eR224-R226

Affie, 2020-02-17:
So this issue becomes:
Add validation to `updateFactor` to error if variables are different. ` error("Cannot update the factor, the neighbors are not the same.")`
- [ ] GraphsDFG
- [ ] LightDFG

GearsAD, 2020-02-17:
Sounds like a plan - like the idea that we check that - links and their order are immutable, so good to fail.

Affie, 2020-05-13:
It would be safest to make `_variableOrderSymbols` immutable. In the In-Memory drivers it is possible to change the variable order since its by reference.

closed 2020-05-14 by Affie