JuliaRobotics/DistributedFactorGraphs.jl #300: Refactor graph copy functions

Affie, 2020-02-14:
Look at draft of  `copyGraph` functions in attic/sandbox.jl
The idea is to have only one function to be called by functions such as `buildSubgraphFromLabels`

Will at least look at it for v0.6.0 and decide if it should be added.

Affie, 2020-02-21:
related to #250
Will refactor and consolidate in the v0.6.x cycle

dehann, 2020-02-28:
agree, `buildSubgraphFromLabels` should be using common DFG functions to minimize code maintenance.  Might need a keyword or two for more use cases.

Affie, 2020-03-30:
### Current state
I don't quite get the there are currently 5 subgraph functions comment, but lets look at all related functions:

Currently we have `_copyIntoGraph!` as the core function.
- `getSubgraph` (basically an alias for `_copyIntoGraph!`)
- `getSubgraphAroundNode`  (`getNeighborhood` -> `_copyIntoGraph!`)
     - can possibly be consolidated with `getSubgraph`
     - Maybe use `build` verb (`buildSubgraph`) and then use multiple dispatch for "AroundNode" part.

We then  have 2 "IIF" functions, but I think its mostly because of a misunderstanding.
- `buildSubgraphFromLabels!` (I don't think this function should exist at all, but I guess that's why we have 2)
- `buildCliqSubgraph!` (This doesn't use `_copyIntoGraph!` yet, but its planned with its replacement `deepcopyGraph!`. see snipped bellow.

`buildCliqSubgraph!` builds a very unique subgraph with frontals and separators handled differently (so I believe it has a place as its own function in IIF). Part of the motivation is to replace the "copy" with a "view" in the future, ie. `viewGraph`.  
```julia
function buildCliqSubgraph!(dfg::AbstractDFG,
                            cliqSubFg::AbstractDFG,
                            frontals::Vector{Symbol},
                            separators::Vector{Symbol};
                            solvable::Int=0)

  allvars = union(frontals,separators)

  #get list of factors to possibly add, ie. frontal neighbors
  addfac = Symbol[]
  for sym in frontals
    union!(addfac, getNeighbors(dfg,sym))
  end

  #filter to only the factors in the clique
  allfacs = Symbol[]
  for sym in addfac
    vos = getVariableOrder(dfg, sym)
    if vos ⊆ allvars   
      union!(allfacs, [sym])
    end
  end

  # add all the factors and variables to the clique subgraph
  deepcopyGraph!(cliqSubFg, dfg, union(allvars, allfacs))

  return cliqSubFg
end
```

### Summary
I would suggest and will implement if there are no objections:
- A core copy function. `_copyIntoGraph!` improved and renamed to `copyGraph!` with "aliases" such as `deepcopyGraph!` and `Base.convert`. Drivers can overwrite this function for efficiency. see https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/commit/569c3a3af4169ec6c5cd821fc984772f7257738a
- DFG function `buildSubgraph` generic subgraph helper methods for neighbourhoods etc. will call `copyGraph!` to do the work with labels created from neighbors and labels.
- IIF specialized  `buildCliqSubgraph!` function that builds a subgraph from frontals and separators.

PR #373 (2020-04-13): Refactor graph copy and subgraph functions
Close #300
Still to do:
- [x] add tests for `buildSubgraph`
- [x] deprecate `_copyIntoGraph` in favor of `copyGraph`/`deepcopyGraph`/`deepcopyGraph!`
- [x] deprecate `getSubgraph` in favor of `buildSubgraph`
- [x] deprecate `getSubgraphAroundNode` in favor of `buildSubgraph`
closed 2020-04-13 by Affie