JuliaRobotics/DistributedFactorGraphs.jl #302: Fix docs for v0.6

Affie, 2020-02-14:
- [ ] Missing doc-strings
- [ ] Broken links and doc-strings
- [ ] Struct accessor names
- [ ] General updates

dehann, 2020-02-17:
Hi @Affie , what help do you need for General Updates?  You mean read all the docs and make sure they are v0.6 complaint?

Affie, 2020-02-19:
Small fix to docs #314 
If you quickly go though it and its ok we can close this. We can continuously improve as time goes on. Just want to make sure there are no major issues.

dehann, 2020-02-19:
I think we must absolutely have the `!` in cases like this (#315 ):
https://juliarobotics.org/DistributedFactorGraphs.jl/latest/func_ref/#DistributedFactorGraphs.setTimestamp-Tuple{DFGVariable,Dates.DateTime}

Affie, 2020-02-20:
>I think we really must have the ! in cases like this (#315 ):  

no, nothing is modified.

closed 2020-03-03 by dehann