JuliaRobotics/DistributedFactorGraphs.jl #306: FileDFG should just use tar files (deprecate folders)

GearsAD, 2020-02-16:
This would simplify the code and would lead to less user issues.

dehann, 2020-02-16:
another thing I ran into is when the tar gets unpacked, it drops the folder in common location like /tmp/caesar/random, which is a problem if two processes are untarring files with the same name.  Maybe just untar to unique folder in common location instead?

There should be code in there to delete the untarred folder after loading the fg object, otherwise the number of files just becomes crazy high in a short amount of time.

GearsAD, 2020-02-16:
Yep, we should also consider [Tar.jl](https://github.com/JuliaIO/Tar.jl) for tarring/untarring.

dehann, 2020-04-27:
I fixed the common location thing at some point with temporary uuid folder names.

closed 2020-04-27 by dehann