JuliaRobotics/DistributedFactorGraphs.jl #308: Does generated == replace the compare functions?

GearsAD, 2020-02-16:
Does this https://github.com/JuliaRobotics/DistributedFactorGraphs.jl/blob/master/src/services/CompareUtils.jl#L21-L28 replace the compare functions below?

Affie, 2020-02-17:
In my mind, no. I don't think we should change the julia definition of ==. Think compare can still be used for skipping and specialized test? I left compare untouched as it is what IIF uses for tests.
However, I like the skipfield idea, but maybe as a keyword argument. We can also consider isaprox for iif tests.

dehann, 2020-02-17:
so how about these two uses:
- A == B  ; and
- ==(A,B, skip=[:bob])

Affie, 2020-02-17:
I would then suggest API of == and `isapprox`/`≈`  with both ignore.
following julia on `isapprox` but with ignore added.

GearsAD, 2020-02-17:
:+1: for == ignore.

We just use the ignore for tests at the moment and I don't want us to break general use-cases - so it should be fine to just use the ==(..., ignore=[:adam]) there.

dehann, 2020-02-17:
so how about then also having:
- `≈(A, B, ignore=[:thing])`

dehann, 2020-02-17:
this may influence #285

dehann, 2020-02-17:
xref #307

Affie, 2020-02-17:
Also verbose flag for listing the fieldname it fails on.

dehann, 2020-02-17:
Keep in mind @Affie  built this with `@generated` macro, and I understand the fieldname only during verbose is easiest (no fancy messages).

dehann, 2020-02-17:
We need verbose to fix #293

closed 2020-02-21 by dehann